PX4/PX4-Autopilot #7526: How to read queued topics?

lukegluke, 2017-07-04:
Hi all,

I'm trying to understand how to work with queued topics (that is advertised with orb_advertise_queue or orb_advertise_multi_queue) - how can I copy out several messages from queue?

For example, let say gps driver publishes to topic gps_dump data of NMEA messages from 10Hz gps receiver. gps_dump.msg data array is limited by 79 bytes, so in fact messages are published in topic with higher then 10Hz rate (the nmea strings for every measurement is about 200 bytes). The size of queue of gps_dump topic is 8.

Now my module subscribes on gps_dump topic and limit polling rate to 10Hz. On every update I would like to read out all messages that are currently in the queue, something like this:
```
if (fds[0].revents & POLLIN) {
	struct gps_dump_s report;
	uint8_t buf[79*4];			//up to 4 reports
	ammountR = 0;
	bool updated = true;

	//accumulate several reports in one message
	do {
		orb_copy(ORB_ID(gps_dump), gps_dump_sub_fd, &report);
		memcpy(buf+ammountR, (uint8_t*)report.data, (uint8_t)report.len);
		ammountR += (uint8_t)report.len;
		orb_check(gps_dump_sub_fd, &updated);
	} while ((updated) && (ammountR < 79*3));
	...
```
but every time I read out only one 79 bytes long last message. This leads to loss messages (according to uorb top -a).
And according to documentation of **orb_check** it works correctly, cause:
> [orb_check] check whether a topic has been published to since the last orb_copy.
and, yes, it hasn't.

Of course, I can uprate polling rate of my module and it works, but I didn't understand why orb topics can be queued [at least more than 2] at all then if there is no way to read all messages in stock?
I think that I missed something.

Thanks in advance for help!

bkueng, 2017-07-05:
From what I see your understanding is correct and it should work the way you wrote it. Can you show the full code? Do you use `orb_set_interval`?

lukegluke, 2017-07-05:
> Do you use `orb_set_interval`?

Yes, I do.

Here is extended code:

```
/* subscribe to gps dump topic */
int gps_dump_sub_fd = orb_subscribe_multi(ORB_ID(gps_dump),0);
/* limit the update rate to 100 ms */
orb_set_interval(gps_dump_sub_fd, 100);

/* wakeup source(s) */
px4_pollfd_struct_t fds[1];

fds[0].fd = gps_dump_sub_fd;
fds[0].events = POLLIN;

int error_counter = 0;

do {
	/* wait for update for up to 1000 ms */
	int poll_ret = px4_poll(fds, (sizeof(fds) / sizeof(fds[0])), 1000);

	/* handle the poll result */
	if (poll_ret == 0) {
		/* this means none of our providers is giving us data */
	} else if (poll_ret < 0) {
		/* this is seriously bad - should be an emergency */
		if (error_counter < 10 || error_counter % 50 == 0) {
			/* use a counter to prevent flooding (and slowing us down) */
			PX4_ERR("ERROR return value from poll(): %d", poll_ret);
		}
		error_counter++;
	} else {
		if (fds[0].revents & POLLIN) {
			struct gps_dump_s report;
			uint8_t buf[79*4];			//up to 4 reports
			ammountR = 0;
			bool updated = true;

			//accumulate several reports in one message
			do {
				orb_copy(ORB_ID(gps_dump), gps_dump_sub_fd, &report);
				memcpy(buf+ammountR, (uint8_t*)report.data, (uint8_t)report.len);
				ammountR += (uint8_t)report.len;
				orb_check(gps_dump_sub_fd, &updated);
			} while ((updated) && (ammountR < 79*3));
			...
	}
} while (!_thread_should_exit);
```
ammountR is always 79.
If I set orb interval to 5 ms it works (doesn't loose messages)

bkueng, 2017-07-05:
Just remove the `orb_set_interval` call completely, you don't need it for your use case. The rate limiting works on top of the queue, working against for what you're actually trying to achieve.

lukegluke, 2017-07-05:
> The rate limiting works on top of the queue, working against for what you're actually trying to achieve.

Ok, I see. I'm not doubt it will work similar to 5 ms interval.
Just in case, so if I want to limit my module activity to ~10Hz, it should just sleep for 100ms (or as alternative polling on another topics with 100ms timeout) and accumulate several gps_dump reports from orb queue with loop I wrote, right?

bkueng, 2017-07-05:
> Just in case, so if I want to limit my module activity to ~10Hz, it should just sleep for 100ms (or as alternative polling on another topics with 100ms timeout) and accumulate several gps_dump reports from orb queue with loop I wrote, right?

Yes. Just make sure that the queue size is large enough then.

lukegluke, 2017-07-05:
Thanks for help!

closed 2017-07-05 by lukegluke