PX4/PX4-Autopilot #7527: How to get global yaw with vision heading estimation enabled

Changliu52, 2017-07-04:
Hi,

I am using 1.5.5 standard EKF2 firmware,
The related parameter setting is 
EKF2_AID_MASK = 26 (vision estimation, vision heading, optical flow, gps enabled)

In this case, the local_pose was initialised based on the vision quaternion, which always starts facing north from the SLAM module. However, I still need the global yaw for GPS integration. 

I had a brief read on [https://github.com/PX4/Firmware/pull/2667](url), where 

> Mode 2: Use both external heading and magnetometer. Useful for having a correct global yaw for use cases like mine. This still needs much work, and I will remove it from this PR if it holds up the primary integration.

described exactly what I need. But seem not been implemented yet. I hope I missed something, and there are implementations for this mode? Or if not, maybe any suggestions for the quick way around it?

Thank you in advance.
Kind Regards,
Chang

dagar, 2017-12-06:
Were you able to figure this out?

Changliu52, 2017-12-06:
I don’t think this is supported in px4. I implemented heading alignment between vision and mag externally, and realised it is not a trivial problem.

mhkabir, 2017-12-06:
This is now supported. We implemented frame alignment for vision-GPS fusion in EKF2. It will be in the v1.7 Release.

dagar, 2017-12-06:
Thanks for the update Kabir.

closed 2017-12-06 by dagar