PX4/PX4-Autopilot #7533: how can I generate IR or bytecode for each source code file?

Tigerly, 2017-07-05:
I'm trying to generate call graph (need to compile and generate IR or bytecode first) for the PX4 source code by using LLVM clang.
e.g.,
..PX4/Firmware/src/modules/navigator$ clang++ -S -emit-llvm land.cpp -o - | opt -analyze -dot-callgraph

However, it seems the source code files cannot be compiled normally.
It always says "land.cpp:47:10: fatal error: 'systemlib/err.h' file not found #include <systemlib/err.h>"
if I change this header file's path, it will complain other header files.

My question is:
How can I generate IR or bytecode for the PX4 souce code (e.g., land.cpp in PX4/Firmware/src/modules/navigator)

Thanks in advance!

dagar, 2017-07-05:
Try it within the SITL posix build.

`make posix_sitl_default`

Then within the build directory you can see how land.cpp is actually built in `build_posix_sitl_default/compile_commands.json`.

dagar, 2017-07-06:
Please reopen if this didn't answer your question.

closed 2017-07-06 by dagar