PX4/PX4-Autopilot #7535: Flyaway after loss of datalink (v1.6.2)

BazookaJoe1900, 2017-07-06:
Hello,
Few days ago I flown my multicopter for about 1.5 km. and almost got flyaway, only using RC saved it for safe landing. 
I got 'ALL DAT LINK LOSS' and it seems that RTL has been engaged (I still had downlink). but, instead of flying toward home position for landing the multicopter started to fly very fast backward.
from the log I see that just after losing the datalink (at 10:28) it did got new heading, X and Y setpoints of home. but just after that (when the link was regained?) the position setpoint sets back to its current position, no change of heading setpoint. and the multicopter gets higher and higher pitch, that leads it to fly very fast backward, not closing the X and Y position setpoints. I have tried to send new commands reposition, pause the multicopter seems to be affected by them but still contiues to fly at very high speed.
At this point (2km away) we toke RC control, and fly it to home for landing.

log: 
http://review.px4.io/plot_app?log=2da94d7c-a374-4bb3-b7f9-efd8fecaf445

dagar, 2017-07-06:
@Stifael is this a duplicate of https://github.com/PX4/Firmware/issues/7467?

BazookaJoe1900, 2017-07-06:
@dagar and @Stifael. it might be related to #7467, but not excally the same.
In this case there was no command to PosCtl (not by me, maybe automonosly), but yet the setpoints goes to very high value 430m/s and seems to go down
[image]

nkhoit, 2017-07-06:
This is a duplicate of #7501 which is related to #7467 
The problem comes from the same place in mc_pos_control

Stifael, 2017-07-06:
a quick fix here: the main change is that the magnitude for the velocity sepoint has to happen right before the velocity setpoint is constraint, and not somewhere else in the code. In addition, the constraint needs to happen at the end.

BazookaJoe1900, 2017-07-07:
Do you think that #7538 solves also the phenomena of flying backward?

LorenzMeier, 2017-07-08:
Yes. And yes, there will be a fix release shortly.

BazookaJoe1900, 2017-07-08:
@LorenzMeier  Thanks!
I can recommend, if it possible, to release the 1.6.4 with only fixes to issues (like this) in order to avoid inserting other "features" that might have side effects, bugs, breaking other features and protocols.

LorenzMeier, 2017-07-08:
That's our standard mode of operation. We haven't done more than fixes since v1.6.3.

LorenzMeier, 2017-07-10:
Closing this, as it is contained on master.

closed 2017-07-10 by LorenzMeier