PX4/PX4-Autopilot #7537: wrong cruise speed when continuing mission

ChristophTobler, 2017-07-06:
When setting up a mission with a non-default flight speed, the cruise speed will change to -1 when continuing the mission.
[image]
Result is a different flight speed.

dagar, 2017-07-06:
At what point did you set the cruise speed, and then when did it change?

ChristophTobler, 2017-07-06:
At the beginning. It changed once I continued the mission after I interfered by switching to pos ctrl (~38s)

dagar, 2017-07-06:
I think that's actually the intended behavior. The change speed command is executed once, but any interruption resets cruise to default. I wonder if some mechanism to save it per mission would make more sense.

Stifael, 2017-07-06:
Would be nice to save the previous setpoint in general when stopping and restarting the mission

dagar, 2017-07-06:
We could keep it for Mission and only reset if when a mission actually changes (uorb offboard_mission onboard_mission). 
If we only want cruise speed (includes COMMAND_LONG) to apply in mission this becomes easier. Anyone feel strongly one way or the other?

closed 2018-02-21 by PX4BuildBot