PX4/PX4-Autopilot #7539: Run CI with gcc 7

jlecoeur, 2017-07-06:
Starting from version 7, gcc provides additional warnings, see here for the detail https://gcc.gnu.org/gcc-7/changes.html.
In PX4, these warnings were fixed for posix targets in #7332, and arm targets in #7502. 
The benefits are:
- users of distros with recent versions of gcc and arm-none-eabi-gcc (such as archlinux) can happily compile PX4 using the default compilers
- more warnings means safer code, for instance this bug was difficult to notice but the compiler catched it: https://github.com/PX4/ecl/commit/d0b4f1e225a5252d543cbc419d642277df535ee7. The implicit-fallthrough warning is also nice because it imposes to explicitly document missing breaks in a switch statement.

But the compatibility with gcc 7 needs to be enforced by CI. @dagar proposed to add a docker image
> If someone would like to get a gcc 7 docker image together I'll get it into the build system.
https://github.com/PX4/containers/blob/master/docker/px4-dev/Dockerfile_nuttx

However I am not familiar with docker. I just found this Dockerfile that might do the job for posix targets (see https://hub.docker.com/_/gcc/ and https://github.com/docker-library/gcc/blob/3b33871fe9558262cb5ed6253d358f76710e9ccb/7/Dockerfile).
Anybody willing to help is welcome!

dagar, 2017-07-06:
Building GCC seems a little excessive. I'll try to find a package.

LorenzMeier, 2017-07-18:
@dagar Any news?

dagar, 2017-07-19:
I have px4-dev-base with Ubuntu 17.10 (GCC 7 default) tagged here - https://hub.docker.com/r/px4io/px4-dev-base/builds/
I'll add a GCC 7 SITL build initially, and then we can look at adding arm.

PR #7644 (2017-08-10): travis-ci add GCC7 SITL build and tests
 - closes #7539 

@jlecoeur once this is passing the GCC 7 SITL build is working.
commit 02cee074 (2017-08-10): travis-ci add GCC7 SITL build and tests

-closes #7539
closed 2017-08-10 by dagar