PX4/PX4-Autopilot #7544: Unexpected back transition

LorenzMeier, 2017-07-06:
From flight test team, FYI @dagar http://logs.px4.io/plot_app?log=78d033a5-d3e8-4e18-a73f-3602f21e78ab

@bresch Can you please have a look as well?

dagar, 2017-07-07:
It was initiated by a brief loss of RC.

[image]

The vtol controller directly handles the transition switch. It should probably be centralized with the rest of the switch handling.

LorenzMeier, 2017-07-07:
I think its architecturally ok. The manual control state should not update when RC is lost.

dagar, 2017-07-07:
We don't log manual_control_setpoint, but in this case the RC inputs changed before the signal_lost was set. Is this just the configuration of the RC receiver?
@santiago3dr

[image]

LorenzMeier, 2017-07-08:
RC loss.

closed 2017-07-08 by LorenzMeier