PX4/PX4-Autopilot #7545: PX4 1.6.3: Strange problem with Airspeed-less transition.

tubeme, 2017-07-06:
We were flying 1.6.3 for some time. But recently decided to reset all the parameters with the new defaults. For this we use change of air-frame procedure. So we carefully introduced our PIDs on top of the new defaults. We removed the circuit breaker for the airspeed and thus it did not ask us to calibrate the airspeed so the diff press cal params were empty 0.00. 

In this setup we were not able to complete the transition. It never stopped the quad despite the speed got exceeding 20 m/s and the transition continued forever up to the crash.

http://review.px4.io/plot_app?log=b6653ab1-ada7-4711-abaf-00a6a51249dc

Because we flew the 1.6.3 airspeed-less without a problem we just started thinking. We enabled the airspeed. Calibrated the airpseed. Then again we disconnected it from the breaker.

On the Bench test perfect transitions.

http://review.px4.io/plot_app?log=619b0b6b-ae84-4220-aa52-73ddce2b7fcc

http://review.px4.io/plot_app?log=3647be91-b6f2-4be1-a5a3-f66ca93c8b94

Also we flew without airspeed without any problem. So the only thing we had to do was to calibrate the airspeed in order to make the transition in aispeed-less config.

During this tests we discovered another thing. In airspeed-less mode the only way we could manage the transition is timing. 

There is a parameter VT_F_TR_OL_TM which is supposed to guarantee fixed transition time without airspeed. But in reality in airspeed-less mode the transition is done now by the VT_TRANS_MIN_TM which is the minimum front transition time. The VT_F_TR_OL_TM did not have any effect over the transition timing.

sanderux, 2017-08-09:
Please follow the guide: http://px4.io/docs/vtols-without-airspeed-sensor/

closed 2017-08-09 by sanderux

raygamero, 2017-09-18:
https://docs.px4.io/en/config_vtol/vtol_without_airspeed_sensor.html is the new location of the guide.