PX4/PX4-Autopilot #7547: PX4 1.6.3: Position Takeoff and Land problems

tubeme, 2017-07-07:
Hello guys,

After flying for some time with the 1.6.1 and 1.6.3 we noticed some corners that can be polished regarding taking off and landing in Position mode.

The feature of activating auto take off when in Position mode with the throttle lever passing 50 percent. There is nothing wrong with the 1.5 m/s take off speed, but it happens so sudden that it is scary sometimes.

Now the current take off throttle curve looks like this:

[image]

So this sudden kick of throttle is bothering sometimes. OK we understand that we have to take precautions for the turbulance on the ground and thus take off faster rather than slower. But it it could be fast take off as well 1.5 - 2 m/s but not with kicks but rather smooth throttle curve and thus smooth movement upwards. Like this:

[image]

The Landing is OK thus far. But there might be more improved version for the VTOL. We have wings and on the ground if the weather is strong we might have some movement of the frame on the grouns. No vertical movement but rather shaking side to side. So this tricks the land detector by not detecting landed state.

dagar, 2017-12-06:
Sorry Vasil, this slipped through without a response.
Has any of this been improved by v1.7.0RC?

RomanBapst, 2018-03-06:
@MaEtUgR @Stifael This is not the case anymore, right? I think we can close.

Stifael, 2018-03-07:
> So this sudden kick of throttle is bothering sometimes. 

I think this is fixed with the throttle ramp. However, @RomanBapst noticed [here](https://github.com/PX4/Firmware/issues/8885), the ramp can still be farther improved.

> But there might be more improved version for the VTOL
... movement but rather shaking side to side

Agreed. I think the landing is now quite robust given the parameters are not totally set incorrect. And even then, there are now three stages which ensure that false-positive detection is basically not possible. However, I also know that for vtol certain parameters are kept large for whatever reason (such as minimum thrust), which makes the detection more vulnerable. 

I think for now we can close it. That should not mean that the detection is perfect, but the detection is currently at a state where it is acceptable.

LorenzMeier, 2018-03-07:
Thanks @Stifael !

closed 2018-03-07 by LorenzMeier