PX4/PX4-Autopilot #7548: PX4 1.6.3

tubeme, 2017-07-07:


dagar, 2017-07-07:
Can you summarize how you'd like it to work?

tubeme, 2017-07-07:
Two ways simple and deeper:

Simple way.

Make Flaps fully extend in Manual modes and Guided modes and Mission.
Make the Flaps extend slowly in all modes like it is in Position mode now.
Provide a parameter to control the timing of the flaps extension. IE 2 seconds for the movement from current position to the set point.

Deeper way.

Give us a scaling parameter for manual modes. We decide how much the FLAPS will extend in Manual modes.
Give us a different scaling parameter for guided modes Alt/Pos and Mission. We decide how much the FLAPS will extend in Guided and Auto modes.
Give us parameter to control the timing for Manual Modes, so we could decide how fast or slow the FLAPS will perform in Manual modes.
Give us another parameter to control the timing for Guided and Auto modes so we could decide how fast or slow the FLAPS will perform in Guided and Auto modes.

dagar, 2017-07-07:
Thanks for the detailed response.

For manual mode do you need flaps to be binary (ON/OFF), have multiple positions, or be continuous (RC knob)?

Why is it important to control the timing? Would a longer default be sufficient? Maybe the deploy speed should actually be a function of airspeed?

Why do the flaps need to behave differently in alt/pos, mission, etc?

I'm trying to understand fundamentally how they need to work. If possible I'd like to avoid adding many tuning parameters when those same values can be derived.

tubeme, 2017-07-08:
No binary FLAPS. It is always a good idea to have control over the extension. If I use 3Pos switch then I have  two positions for the flaps in strong wind we use middle extension in good conditions we need full extension. If I use rotary knob then that is OK as well we have control all the way. Different people different styles.

The timing is needed because there are different types of aircraft and different flaps. We need fast (but no sudden) flaps for parachuting to the ground (flaps that extend almost perpendicular to the wing) We might need 0.5 - 1 sec extension. While for other aircraft (larger one that land on a tarmak) we might need something like 2-4 sec. So PX4 is kind of universal and working with different machines we need more control over the flaps that is appropriate for the machine. If we go in the direction of deployment corelated to the airspeed, we will create a new mess we will have to tune it, test it etc etc. and at the end this corelation will be different for the different frames again we will need some other sort of parameter tuning to make it work properly. Because Flaps are set and forget once per setup thing it is better to be as simple as possible. So couple of parameters will do a perfect job. The guy who makes the machine will tune it to their frame and have it working all the time "not dependent on faulty airspeed sensor" for example.

Why do we need scaling. The thing with flaps is that they are some problems when you tune them. Usually the 0 point has to be perfectly aligned with the wing. If the mechanic did not implement adjustable rods, everything should be properly tuned in the RC. So you manually tune the 0 point such that the flap is aligned with the wing. The same holds for the fully extended position. Very often the full move of the servo is too much flap. So we have to reduce the movement ie we do trimming in the RC. Instead of doing it in the RC it is good idea to make it in the PX4, because otherwise the FLAP becomes dependent on the RC. Which is not good idea if we are RC'less flying. And also because it will not be possible to automate it in the Mission etc.

Why do we need different sets for Manual and Pos/Alt + Mission. There are only two sets. One for the manual modes one for the guided and auto modes.

Sometimes we might want to be more aggresive with the flaps for some reason. This could be the case in Manual. When you are in manual it is supposed that you know what you are doing so you might need more aggresive timing and extention for the flaps. While in the guided modes +auto modes you have to be more gentle with the flaps because there is certain autonomy of the machine so we cannot go to the extremes. This means that this way we will have two whole sets of flap parameters. Triming + timing for the manual (more aggresive if we need it) and Triming + timing for the guided and auto modes (more gentle if we need it).

Of course there could be just one set of parameters triming + timing but then dependent on our needs we might loose the option to differentiate the flaps in the two modes.

I know that it is good idea to avoid parameters in general. But very often it proves that adding two or four parameters is much better, much simpler and more reliable than the derived option. Like I mentioned before what will happen if we are in airspeed'less mode?? How do we extend the flaps? Or if we have a faulty reading of the airspeed?

Lastly Flaps are set and forget thing. You tune it once you forget about it until the plane is dead. But we need some tools to properly tune it. Even if 90% of the people will not use this tuning in both modes, there might be some 10% that will need it. And it is better to have it and not use it instead of needing to set it up but not have the option. That makes the great product great, not the 90% satisfaction. But this extra 10% of detail will bring a lot of happy smiles.

dagar, 2017-07-08:
It doesn't mean it has to be a dumb function of airspeed. Flaps increase the lift coefficient (and drag) and lower the minimum airspeed. For example this could be yet another tool the FW position controller (TECS) has at its disposal (with or without an airspeed sensor), but the utility decreases if you force it into an artificially simple model. You could then have a lower minimum airspeed configured (FW_AIRSPD_MIN) than you could achieve without flaps, but the controller would deploy them incrementally only as needed.

If the parachute/airbrake scenario is important we should do it explicitly instead of trying to deceive the autopilot by having it deploy flaps quickly. Tell it what you want to do, don't try to trick it into achieving what you want.

With scale factors we're just replicating functionality that's already in the mixer. If it's too hard to do it in the mixer then we should focus on making that easier instead of adding even more configuration on top.
If you want to adjust the center of a servo there are already per PWM parameters for offsets.

The main thing I'm wondering for manual is you wanted to manually control the flaps (RC PWM -> FLAPS) or simply ask for position 1, 2, etc. We could simply slew it, but I'm now wondering if we should be doing anything at all. In many (most?) transmitters you can set a delay. If you really want manual mode perhaps the autopilot should stay out of it entirely.

For now I'm happy to make the hard coded flap deployment time a parameter, I just want to pause and consider both short term needs and potential longer term solutions.

tubeme, 2017-07-10:
@dagar I agree with your arguments. Will think about the use case scenarios and write them down. Then we could move  on. 

I agree with the scaling and the Mixer. But last time when I tuned the throttle of an IC engine servo I spent 1 hour tuning it with the current mixer, and spent 20 minutes for the flaps. 

And I don't know a person here that understands what is going on in the mixer and can edit it. So you are correct probably we need a decent Mixer editor... I remember there was a PR for this. Even @DonLakeFlyer created a Tab in the Parameters view. The Mixer indeed is very powerful, but with cumbersome control. 

PWM parameters for offsets are not usable. To enable this by channel PWM parameters I have to create a custom script otherwise they apply to the whole bus. I have a trim per channel from the Parameters view, but this is not what we need because we need to control the throws as well to tune the flaps. The same holds for the throttle servo if we use IC engine. Probably a decent Mixer editor could be great to solve all this problems.

closed 2018-02-21 by PX4BuildBot