PX4/PX4-Autopilot #7550: EKF2 FW defaults

dagar, 2017-07-07:
@priseborough should we set different EKF2 defaults for FW (or MC, VTOL)? We have a mechanism to set different initial values for each platform type if needed.

Should airspeed fusion be enabled by default for FW? What about synthetic side slip?

    EKF2_ARSP_THR 0
    EKF2_FUSE_BETA 0

Would it be safer to disable the default magnetometer fusion method switching for FW?

    EKF2_MAG_ACCLIM  0.5
    EKF2_MAG_YAWLIM 0.25

Any others? The defaults are here. https://github.com/PX4/Firmware/blob/master/src/modules/ekf2/ekf2_params.c#L42

dagar, 2017-07-07:
Default ublox 4g dynamic model?

priseborough, 2017-07-07:
I believe we should be using the 4g dynamic model as the default to handle possible catapult launches rather than waiting for users to discover the need for themselves given that the downside for fixed wing is just a bit more GPS velocity noise which the EKF can easily cope with.

Setting  EKF2_MAG_ACCLIM  = 0 and  EKF2_MAG_YAWLIM =0 is also prudent.

Enabling airspeed fusion provides a wind estimate and adds dead-reckoning ability within the EKF if GPS is lost, however after reviewing the code I do think the control loops are able to use the estimates because of the way the global position message is published by  ekf2_main.cpp. This is something I will fix in the next few days. 

At this stage the only code using the wind estimate published by the EKF is the bottle drop calculator, telemetry and logging.

A threshold value for EKF2_ARSP_THR of 8 m/s would handle the requirements for most planes.

Enabling side fusion enables wind estimation when there is no airspeed sensing and also improves the wind estimate and dead-reckoning performance, however it comes at the cost of slightly larger velocity innovations and can potentially cause GPS rejection if the vehicle is flown in aerobatics involving with large sideslip. For this reason we had better use EKF2_FUSE_BETA = 0 as the default until we are confident it can handle all types of manoeuvres and leave it as a specialised setting for applications where a non airspeed sensor wind estimate is required. I need to update the wiki to include advice on setting this parameter.

PR #7560 (2017-07-08): FW EKF2 and ublox defaults
 - closes #7550
commit 75a528e0 (2017-07-08): FW EKF2 and ublox defaults

 - closes #7550
closed 2017-07-08 by LorenzMeier