PX4/PX4-Autopilot #7551: PX4 1.6.3: Constant "High Airpseed" errors.

tubeme, 2017-07-07:
Hello guys,

Here is another issue we have since 1.6.0 RC1. We constantly have problems calibrating the airspeed sensor. It does not matter which airspeed sensor we use. 

But depsite that once calibrated the airspeed stays at ~0 m/s The moment we restart the Pixhawk and connect again to the QGC we get High Airspeed error and a RED light and not able to arm. Then we check in the Fly view that the airspeed stays at ~0 m/s. Then we produce airflow in the pitot for about 1-2 seconds and the RED light disapears and it becomes Blue Hearbeat. We are able to arm without a problem. 

It seems to us that by blowing in the pitot we remove the high airspeed error. 

We can reproduce everytime.

Currently using HKPilot32 + FMUv2 1.6.3 and MS4525 sesnor for the tests.

This High Airspeed errors are really annoying. Especially if there is no high airspeed measured.

dagar, 2017-07-07:
It's poorly worded and has been improved recently. The problem is likely a large negative differential pressure, but the system won't show airspeed below 0.

Can you repeat this in log? Set SDLOG_MODE 1 to log from boot until disarm. Let's first verify we understand exactly what's happening, then discuss possible solutions.

tubeme, 2017-07-08:
I will do this later logging at night when my work day begins :). But in the mean time how is it possible to have just made a calibration, and then on a restart to have a large negative differential difference?

I understand that there might be different conditions, but we are talking about 1 minute restart... and it starts with "High Airpseed check calibration" I work with at least 4,5 different aircraft in a different projects, some hobby projects some pro projects. And this is the MOST ANNOYING thing that everybody I work with is pissed off.  I'm fed with it as well. Also we really struggle some times to make the airspeed calibration for 5-6 times in a row...

Also I don't understand how how is it possible that I get the High Airspeed warning, I get the red light, and then I turn the lite Blue and remove the error by just blowing in the pitot for 2 seconds... After this reading I create the machine just is perfectly operational and I can arm, and the reading of the AS is very good as well throughout the whole flight. It was not like this in the 1.5 versions. The things changed in 1.6 RC versions and are like this up to 1.6.3 stable. 

Do you want me to try the master or something else if there are some changes?

dagar, 2017-07-08:
Yes please try master. It's really helpful if you can open issues as soon as possible with logs so we can track down the problem. PX4 v1.6.0 RC1 has been around for 5 months, but I don't experience this problem across multiple different aircraft so I wasn't even aware of it. Often the hardest part of fixing something is getting enough information to understand the underlying problem.

LorenzMeier, 2017-07-10:
Fixes from @dagar are on master. Please re-open if not fixed.

closed 2017-07-10 by LorenzMeier

tubeme, 2017-07-10:
@dagar @LorenzMeier Indeed no problems in master... This master is a very precious one...