PX4/PX4-Autopilot #7552: Support of GPS setpoints in OFFBOARD mode

mzahana, 2017-07-07:
Hi.
I would like to know if I can send GPS setpoint to PX4 in offboard mode. One work around that I know is to to do some transofrmation to transform the difference between current gps point and target point to a local_ned position setpoint which px4 accepts. However, it would be more convenient to have this directly implemented in px4 firmware.

If not currently implemented, where should it be implemented ? Maybe I can give it a try. Also, a corresponding mavlink msg would be required, right?

Thanks.

TSC21, 2017-07-07:
@mhzana PX4 is currently working torwards having navigation on the local frame only, meaning there is going to be always a correspondence/transformation between global and local frames on the Firmware side. If you are using MAVROS, maybe you will want to try `fake_gps` plugin, which transforms GPS into local coordinates already before sending the setpoint msg to the target FCU.

TSC21, 2017-07-07:
In any case the support you are talking about is also possible by adding an handler for [`SET_POSITION_TARGET_GLOBAL_INT`](http://mavlink.org/messages/common#SET_POSITION_TARGET_GLOBAL_INT) and then the respective usage on the the Firmware.

mzahana, 2017-07-07:
Thanks for the response. The idea is to have the option to send GPS
setpoints in outdoor setup with available GPS sensor. The fake_gps plugin
in mavros,
https://github.com/mavlink/mavros/blob/master/mavros_extras/src/plugins/fake_gps.cpp
, does not send setpoints. It basically sends faked gps data out of mocap
or vision estimates, hence the name fake_gps.

However, what you suggested in the second reply makes more sense.

There are two possible ways to solve this.
1) making a mavros plugin which takes gps setpoints and convert it to local
frame setpoints. Then, sends it to px4. This is easy to do, but will only
work with through mavros.

2) introducing a new setpoint type (e.g. setpoint_gps) with corresponding
maks type and implementation on px4 firmware. This is more complex than
first option, but not restricted to mavros.

TSC21, 2017-07-07:
You don't need to create a new msg as the one I told you is enough. `fake_gps` fits another purpose, you are right, I just wasn't sure what was the source of the data. But a new plugin called `setpoint_global` would make sense, considering that it receives lla data (geographic_msgs can handle data perfectly) and then do the conversion as it is done on `fake_gps` (LLA->ECEF->ENU->NED). You are welcomed to add the support and will gladly review it. The same thing if you bet on support global setpoint on the FCU side, but you should consider add that support for the `global_to_local` branch.

mzahana, 2017-07-10:
I will start with the Mavros option as Ibhad that implemented and tested in flight some time ago. Then, probably later I will look into the FCU implementation.

TSC21, 2017-07-10:
> I will start with the Mavros option as I had that implemented and tested in flight some time ago. Then, probably later I will look into the FCU implementation.

Great. Have a look at `fake_gps` so to know what functions you can use to apply the transform LLA->ECEF->ENU.

TSC21, 2017-07-27:
@mzahana done any progress in this?

mzahana, 2017-07-27:
Will send a PR over the next week.

mzahana, 2017-08-06:
I created a mavros plugin in 
https://github.com/mavlink/mavros/pull/764

TSC21, 2017-08-13:
@mzahana we need an update on the PR so it can be merged. Thank you

mzahana, 2017-12-14:
Now, using https://github.com/mavlink/mavros/pull/764 (already merged into mavros master), GPS setpoints can be sent via MAVROS.

closed 2017-12-14 by TSC21