PX4/PX4-Autopilot #7565: Implement Safe Landing Spots (SLS)

LorenzMeier, 2017-07-09:
We have almost all of the infrastructure in place already. There are only a few minor steps missing:

  * Land should choose the closest landing site if within a certain `NAV_AUTOLND_RAD` radius. The param should default to -1 to disable. It should read the dataman safe landing spots on activation: [code location](https://github.com/PX4/Firmware/blob/master/src/modules/navigator/land.cpp#L74)

That should be enough to configure the rest of the system to failsafe to land and should allow us to cover a lot of use cases already.

magicrub, 2017-07-09:
How is this different than using multiple DO_LAND_STARTs?

dagar, 2017-07-09:
I believe this would be with something like rally points.

@magicrub how do you use multiple DO_LAND_STARTs? In PX4 I only allowed one because having multiple in the same mission didn't make sense to me.

LorenzMeier, 2017-07-18:
@lfelipe Here are pointers how to implement.

LorenzMeier, 2017-10-04:
@bkueng Where is this standing?

bkueng, 2017-10-05:
WIP PR is here: https://github.com/PX4/Firmware/pull/7743. I rebased it, the current state is:
- it's mergable (with some TODO's in the code). The behavior will stay the same if no safe points are uploaded
- I still cannot test it because QGC does not upload the safe points ( @DonLakeFlyer )
- We should list the different use-cases so that we can change/extend the behavior

TSC21, 2018-02-21:
Still open and relevant.

closed 2019-02-13 by stale[bot]