PX4/PX4-Autopilot #7566: Severe horizontal overshots

jianyang1, 2017-07-09:
After seeing unusual behavior in air, i'd reproduce it at SITL using v1.6.3

When pausing the drone during re-position its creates severe overshots.  
its seem that the severity of the overshot is effected by the distance its has already flown, but the pattern is always the same.

[image]

http://review.px4.io/plot_app?log=fbdcf62f-c454-4b98-9d3d-add1913049af

LorenzMeier, 2017-07-09:
Thanks for reporting this! A fix is on its way. It has to do with the slew rate limiting we introduced with v1.6. @dagar We probably want to pull this into v1.6.4 because its so annoying.

LorenzMeier, 2017-07-09:
@Stifael Your last PR requires a bit of sanitisation (look at the diff, it should be obvious that you added unintended changes) but other than that it should enable us to fix this.

Stifael, 2017-07-09:
Which pr?

LorenzMeier, 2017-07-09:
We can discuss on PX4 slack.

jianyang1, 2017-07-10:
@LorenzMeier is v1.6.4 coming out anytime soon?

LorenzMeier, 2017-07-10:
Early this week

jianyang1, 2017-07-23:
hi @LorenzMeier , it seems that this slew rate fix was implement for mission mode only.
when im using Go-to right after auto takeoff or after pausing RTL the behavior remains the same as before with drifting and overshots maneuvers. i guess its has something to do with the mission state machine but im not sure.

Stifael, 2017-07-31:
@jianyang1, the actual fix is still not in.
but yes, the slewrate screws up the mission waypoint tracking.
there is already a solution, but I prefer to get in two other PRs first since it eases the integration for me quite a lot.

Stifael, 2017-09-13:
this issue should now be fixed. if you think otherwise please reopen it

closed 2017-09-13 by Stifael