PX4/PX4-Autopilot #7567: Inconsistent RTL state

jianyang1, 2017-07-09:
When RTL is being activated, instead of changing the position setpoint to Home, its first changes the position setpoint to the place where is where commend from. Only after its reaching that location, the setpoint is changed to Home position.

In Case of de-activate RTL once and than re-activate it again at any point, the setpoint is changing directly to Home position as it should

Produced on SITL using 1.6.3

http://review.px4.io/plot_app?log=843cd12c-8a4c-48c0-b75c-626258cc04bb
[image]

dagar, 2017-07-09:
The first stage of RTL is to climb (if needed) to RTL_RETURN_ALT in place.

You can see what RTL is doing here - https://github.com/PX4/Firmware/blob/master/src/modules/navigator/rtl.cpp#L129

jianyang1, 2017-07-10:
i understand, yet i think reaching to RTL_RETURN_ALT should be enough for changing the set point to home.
its still keep flying towards this irrelevant x,y poition even after climbing to RTL_RETURN_ALT

BazookaJoe1900, 2017-07-10:
@dagar and @jianyang1  just by looking at the plot I see issue, the gap from the position to the target is not "closed" on some points as it should.

Stich123, 2018-01-09:
What about this issue?
I tried to recreate it but couldn't, did it resolve in the latest version? @dagar

BazookaJoe1900, 2018-04-12:
The issue still exist, its not stall...

santiago3dr, 2018-08-01:
this issue was resolved prior to 1.8.0 release
here's the pr that addressed it
#8412

closed 2018-08-01 by santiago3dr