PX4/PX4-Autopilot #7571: PWM_DISARMED not being used

2ndalpha, 2017-07-09:
Hi,

I'm not sure if it belongs here to the issue tracker or to discussions but I'm having problems setting up my flying wing with pixracer.
Everything works expect the motor, when disarmed ESC beeps no signal when armed it beeps "throttle not the bottom".
ESC: HobbyWing Platinum 30A http://milehighrc.com/Manuals/HW-120-PL_Pro.pdf
I've also tried with HobbyKing Plush 40A, no luck.
If the FC would send a signal below 1000, ESC would initialize but I didn't find a way to do it.
Video: https://youtu.be/JtAop2j5NWk
Conf [PX conf.params.txt](https://github.com/PX4/Firmware/files/1133957/PX.conf.params.txt)

Any help is much appriciated.

dagar, 2017-07-09:
Are you sure the ESC is plugged into the correct output? Did you plug it into output 3 or 4?

If you post your log we can see exactly what values are sent? http://logs.px4.io/
You'll probably need to set it to log from boot (SDLOG_MODE 1) and reproduce.

2ndalpha, 2017-07-09:
0 - left servo
1 - right servo
3 - motor
4 - BEC for the rail

Log: http://logs.px4.io/plot_app?log=369a9598-8796-42e0-a796-d2096891d390
Thank you :)

dagar, 2017-07-09:
I wonder if PWM_MIN/PWM_MAX are causing a problem. Can you try restoring them to default?

2ndalpha, 2017-07-10:
I restored all PWM settings to defaults, unfortunately still the same.
Log: http://logs.px4.io/plot_app?log=ad48a8f9-c33d-4b37-9c8a-ee047126f598

dagar, 2017-07-10:
Is the jump from 0us (disarmed) to 1000us (armed) the problem? I thought it should be defaulting to send 900us disarmed, but it might be missing something in the config. Do you know how to access the console? Either physical or through QGC. If so, run `pwm info' and share the output.
[image]

2ndalpha, 2017-07-10:
My biggest issue is I can't get the motor working no matter what.
My thinking was if FC would send proper disarmed signal the ESC would initialize and then it would be good to go.

pwm info:
```
channel 1: 0 us (default rate: 50 Hz failsafe: 0, disarmed: 0 us, min: 1000 us, max: 2000 us, trim:  0.00)
channel 2: 0 us (default rate: 50 Hz failsafe: 0, disarmed: 0 us, min: 1000 us, max: 2000 us, trim:  0.00)
channel 3: 0 us (default rate: 50 Hz failsafe: 0, disarmed: 0 us, min: 1000 us, max: 2000 us, trim:  0.00)
channel 4: 0 us (default rate: 50 Hz failsafe: 0, disarmed: 0 us, min: 1000 us, max: 2000 us, trim:  0.00)
channel 5: 0 us (default rate: 50 Hz failsafe: 0, disarmed: 0 us, min: 1000 us, max: 2000 us, trim:  0.00)
channel 6: 0 us (default rate: 50 Hz failsafe: 0, disarmed: 0 us, min: 1000 us, max: 2000 us, trim:  0.00)
```

dagar, 2017-07-10:
There's something missing in the airframe config. We can use the command line to figure it out and then update.

Try `pwm disarmed -c 4 -p 900` and then `pwm info` again.

2ndalpha, 2017-07-10:
```
channel 1: 1708 us (default rate: 50 Hz failsafe: 0, disarmed: 0 us, min: 1000 us, max: 2000 us, trim:  0.00)
channel 2: 1383 us (default rate: 50 Hz failsafe: 0, disarmed: 0 us, min: 1000 us, max: 2000 us, trim:  0.00)
channel 3: 1500 us (default rate: 50 Hz failsafe: 0, disarmed: 0 us, min: 1000 us, max: 2000 us, trim:  0.00)
channel 4: 900 us (default rate: 50 Hz failsafe: 0, disarmed: 900 us, min: 1000 us, max: 2000 us, trim:  0.00)
channel 5: 0 us (default rate: 50 Hz failsafe: 0, disarmed: 0 us, min: 1000 us, max: 2000 us, trim:  0.00)
channel 6: 0 us (default rate: 50 Hz failsafe: 0, disarmed: 0 us, min: 1000 us, max: 2000 us, trim:  0.00)
channel group 0: channels 1 2 3 4
channel group 1: channels 5 6
```

Still does not work, though. Are ESC-s ok with 50Hz rate?

dagar, 2017-07-10:
To sanity check you definitely have a gap between the servo and ESC? Have you tried this ESC with a regular receiver?

The ESCs I use are fine with 50-400Hz but you could try that as well. Note the channel groups at the bottom. You can only set the rate per channel group.

`pwm rate -c ${PWM_OUT} -r ${PWM_RATE}`

2ndalpha, 2017-07-10:
Those pwm settings should persist, right? After power cycle it is again default value.

dagar, 2017-07-10:
No it's all done in the init partially coming from parameters -  https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rc.interface

Using the pwm command you could also try test to see if you can get anything.

2ndalpha, 2017-07-10:
I got the motor turning by following commands:
```
pwm arm
pwm test -c 4 -p 988
c
pwm test -c 4 -p 1100
```

I've set it up as FX79: https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/mixers/FX79.main.mix
Maybe there is something wrong with it?

2ndalpha, 2017-07-10:
I think I figured out how to make it work. My ESC is calibrated with too low lower value.
If I calibrate it with the lowest value above 1000 it should be fine?
I will try it out tomorrow

r0gelion, 2017-12-14:
@2ndalpha how the test went? is this issue resolved?

closed 2018-08-01 by santiago3dr