PX4/PX4-Autopilot #7572: FW Minimum Airspeed scaling for banked stall avoidance

JonReacher, 2017-07-09:
I think it would be wise to enable scalable minimum airspeed that would increase FW_AIRSPD_MIN according to degrees of bank. I just had a pretty bad crash that, from what I can tell, would have been avoided by having this enabled. The problem gets exacerbated by airframes with higher minimum airspeeds and higher wing loading.

 I know it's possible to just increase FW_AIRSPD_MIN, but that has a ripple effect on other things, including FW_LND_AIRSPD_SC because you can only go as low as 1.0. Although, maybe increasing FW_AIRSPD_MIN and setting FW_LND_AIRSPD_SC to 1.0 is good enough since it will automatically mean having a landing speed buffer without actually having to scale it. The problem with that is that every user has to know that FW_AIRSPD_MIN must not be set to the airframe's actual minimum desired speed in straight and level flight, but they have to know to multiply it by 20-25%.

  The other option is for a user to limit bank angle, but this also has other consequences.

JonReacher, 2017-07-09:
Here's a log from the flight. http://logs.px4.io/plot_app?log=691f09d3-80ae-406b-ac71-6dfca874ee6a

dagar, 2017-07-09:
Hi Jon, I'm interested in this and I'll find some time to look at your log in detail later.

Last year I hacked the FW controller to scale the minimum airspeed based on roll, but never tested it outdoors or tried to merge. https://github.com/dagar/Firmware/commit/b9b7d63b2a8a66a93db412bd5fc469e1fa17e226

I'd to happy to resurrect this if you're able to help with testing. I was also considering limiting the allowable roll angle in the attitude controller based on airspeed.

JonReacher, 2017-07-09:
Yeah, I would happy to test it. The math in your code looks like what's needed. 
The problem with me is that I am not presently any good at coding, compiling, etc. I would need a literal step by step to use anything other than the QGC firmware installer. By not good, I mean completely ignorant.

LorenzMeier, 2017-07-09:
@dagar Let's get a PR up with associated binary!

Antiheavy, 2017-07-09:
I too would love to see an improvement like this. We've been bitten by inside wing stalls in tight turns and have had to increase min speed and decrease bank angle to prevent it. Happy to flight test.

dagar, 2017-07-10:
Pull request - https://github.com/PX4/Firmware/pull/7575

tubeme, 2017-07-10:
4 months ago we've had a tip stall in Position during sharp turn, because we forgot to lower the roll angle from the defaults. We were flying pretty slow around the min airspeed. Probably this scaling could have prevented the crash.

PR #7575 (2017-09-19): FW raise min airspeed based on commanded bank
Needs review and testing.

 - closes #7572

@JonReacher @Antiheavy FYI
I'll share a link to the binary once it's available.
closed 2017-09-19 by dagar