PX4/PX4-Autopilot #7580: IR-Lock

kevinlyu1, 2017-07-10:
Does anyone know how to get the IR-Lock to work with GQC?

I've done all the set up for it using the PixyMon, but after that I can't find anything with integrating it on the px4 stack with the pixhawk. Was wondering what type of parameters I would have to set or does the IR-Lock start to work automatically.

TSC21, 2017-07-10:
@kevinlyu1 IRLock has a driver on firmware but the data of it is no being used still.

kevinlyu1, 2017-07-10:
So basically the IR-Lock won't work on the px4 stack?

TSC21, 2017-07-10:
The driver is here: https://github.com/PX4/Firmware/blob/4453e4201b7a245cff52beeb38a293161aea4c48/src/drivers/irlock/irlock.cpp. From here to a implementation of usability there are some things that can be done (data logging, precision landing, data wrap to `LANDING_TARGET` msg). So the driver allows to get the data on the stack, but nothing is done besides the driver right now. You are welcomed to purpose a PR that implements that usage.

kevinlyu1, 2017-07-10:
What do you mean by a PR?

TSC21, 2017-07-10:
Pull Request

kevinlyu1, 2017-07-10:
Okay thanks!

closed 2017-07-10 by kevinlyu1

ShikherVerma, 2017-10-13:
@TSC21 Hey I am working on px4 and pixy camera using the `irlock` driver included in `Firmware/src/drivers/irlock`.

I was interested in contributing the usability features (data logging, data wrap to `LANDING_TARGET` msg) since I'll be needing them soon.

Can you point me to some existing drivers which I can use for reference to implement these features?

PS - For people working on the irlock. The driver is not compiled by default. You can add `drivers/irlock` to `Firmware/cmake/configs/<target_board>.cmake` along with other drivers. This would give you `irlock` command in the `nsh` terminal of px4.

TSC21, 2017-10-13:
@ShikherVerma check how other distance sensor drivers are implemented for example. But in this case you would want a `landing_target` uORB topic to publish to and use the data published to it to implement a precision landing procedure in `mc_pos_control` for example when LANDING mode is activated (or when a landing setpoint/waypoint is set and the vehicle is on MISSION mode).

ndepal, 2017-10-13:
@ShikherVerma I am also working on getting the IRLock working in PX4. We have modified the driver to publish its data on a uorb message. Currently I am working on implementing precision landing with it.

What are your goals with the IRLock?

ShikherVerma, 2017-10-13:
I wanted to compare landing with IRLock w.r.t. landing with other techniques. So I would have to do the same things as you. Publish its data on a uorb message and implement precision landing with it. Can you link to your commits/branch where you have worked on irlock ?

ndepal, 2017-10-13:
https://github.com/ndepal/Firmware/tree/feature-precland

Consider this very much work in progress. While the estimator is tested a bit on hardware, I have not flown with it on hardware, only SITL.

I intend to open a PR that includes the driver changes, the beacon estimator, and a precision lander next week.
Would very much appreciate testers.

kintzlr, 2018-01-14:
I'd like to try this out with Intel Aero RTF drone.

Is there an overview doc that describes a quick start? For example:
How does the user activate precision landing?
Any specific parameters that need to be set? 
...etc.

Looking forward to it. Thanks.

ndepal, 2018-01-15:
Documentation is here: https://github.com/PX4/Firmware/issues/7580#issuecomment-336496565

The feature is now available in master.

kintzlr, 2018-01-15:
Thanks for the reply, but I think there's something wrong with that link. It refers to this web page.

ndepal, 2018-01-15:
Sorry, here it is: https://docs.px4.io/en/advanced_features/precland.html

kintzlr, 2018-01-15:
Great! Thank you.