PX4/PX4-Autopilot #7590: The buzzer keeps ringing after power on

lvyanxuan, 2017-07-11:
I have burn the latest firmware，but found that the buzzer kept ringing after power on. It won't stop until i press the safety switch. Is there anything wrong with the new firmware?

tubeme, 2017-07-11:
What you mean by the latest?? I just flashed 1.6.4 stable with the QGC and it works perfectly well. Probably the initial loading is a bit slow on power up but other than that works OK.

LorenzMeier, 2017-07-11:
Can you reset the parameters and configure again?

[image]

lvyanxuan, 2017-07-11:
The QGC version is V3.2.0. And the Firmware version is 1.6.4dev. it might also be an error, Because I choose the stable version when i burn it.
I have tried to reset the parameters, but it just keeps ringing.

LorenzMeier, 2017-07-11:
Can you take a video of the setup?

lvyanxuan, 2017-07-11:
Do you mean showing you all the parameters?

lvyanxuan, 2017-07-11:
Oh it was because that i did't insert the SD card...Thank you for your help.

closed 2017-07-11 by lvyanxuan