PX4/PX4-Autopilot #7591: Position Mode for VTOL in MC Tuning.

tubeme, 2017-07-11:
Hello guys,

We have the following problem we could not solve in Position Mode in MC. We tuned our Position XY PIDs such that we have a very tight and accurate Position hold in MC. But there are certain aspects of the Position Control that we cannot overcome.

Here is the Position Control Dynamics in MC when you use the Pitch Stick to move the Aircraft forward.

[image]

While this is very nice and good for the MC it turns out that it is not very logical for the VTOL.

When we make the XY PIDs tight we exaggerate the effects of: Initial Angle and the Airbraking flare the machine is performing from the diagram.

The Initial angle step for VTOL is not logical because we have a whole wing with its drag and aerodynamics. So more logical is to softly pitch up, up to the setpoint angle, and smoothly gain speed.

While we introduced some expo to the stick in the parameters and softened the Initial Angle it is still there and is not very nice to gain speed in Position mode in MC in the initial moment.

The bigger problem is the Airbraking Flare the machine is performing when we release the sticks. It is very dynamic with very high angle and sudden. So given it is not MC but Airplane again with its drag and aerodynamics sometimes this almost flips the airplane. We are using flying wing so it is much more sensitive on the pitch access. 

This Airbraking in the air is very good for the VTOL as well but not at the extend it is happening now. 

I suppose we could lower the XY PIDs to minimize the extend of the flare but are we going to loose the Position accuracy and grip in Position mode in MC?

I could not find a way to control the Initial Angle and Airbraking Flare with parameters. 

Any ideas?

dagar, 2017-07-11:
Try lowering the horizontal acceleration and deceleration.

MPC_ACC_HOR_MAX
MPC_DEC_HOR_MAX

dagar, 2017-07-11:
The other thing you should consider is significantly lowering the max multicopter pitch and roll rates to what you think is reasonable for a wing.

MC_PITCHRATE_MAX
MC_ROLLRATE_MAX

tubeme, 2017-07-12:
We limited the pitch rate which solved the problem in half but the acc and dec are parameter I've missed along the lines.

Thanks a lot @dagar We will try the parameters and see what is going on.

dagar, 2017-07-15:
I'll close this for now. Reopen if you think it's worth changing some of the vtol defaults.

closed 2017-07-15 by dagar