PX4/PX4-Autopilot #7596: Multi-Vehicle Simulation Error

gyang1011, 2017-07-11:
Hello, 

I am following the guide on multi-vehicle simulation: https://dev.px4.io/en/simulation/multi-vehicle-simulation.html. 

But I got an error when I run 
       `roslaunch px4 multi_uav_mavros_sitl.launch`

The error is the following:

```
Done checking log file disk usage. Usage is <1GB.
Usage: xacro.py [options] <input>

xacro.py: error: expected exactly one input file as argument
while processing /home/burobotics/Guang/src/Firmware/launch/single_vehcile_spawn.launch:
Invalid <param> tag: Cannot load command parameter [iris_1_sdf]: command [/opt/ros/kinetic/share/xacro/xacro.py /home/burobotics/Guang/src/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.xacro rotors_description_dir:=/home/burobotics/Guang/src/Firmware/Tools/sitl_gazebo/models/rotors_description mavlink_udp_port:=14560 > iris_1.urdf ; 'gz sdf -p iris_1.urdf'] returned with code [2]. 

Param xml is <param command="$(arg cmd)" name="$(arg vehicle)_$(arg ID)_sdf"/>
The traceback for the exception was written to the log file
```
Note: when I run the single vehicle simulation everything works just fine. The mavros is also working and I can `rostopic echo` the sensor messages. I only have issues with the multi vehicle simulation.

gyang1011, 2017-07-13:
It looks like I cannot load iris_1_sdf in the single_vehcile_spawn.launch. Any idea how to fix that?

gyang1011, 2017-07-13:
OK. I am trying to write my own launch file.
'''
<launch>

    <arg name="debug" default="false"/>
    <arg name="verbose" default="false"/>
    <arg name="paused" default="false"/>

    <arg name="est" default="lpe"/>
    <arg name="world" default="$(find mavlink_sitl_gazebo)/worlds/empty.world"/>

    <arg name="headless" default="false"/>
    <arg name="gui" default="true"/>

    <arg name="pluginlists_yaml" default="$(find mavros)/launch/px4_pluginlists.yaml" />
    <arg name="config_yaml" default="$(find mavros)/launch/px4_config.yaml" />

    
  <include file="$(find gazebo_ros)/launch/empty_world.launch">
      <arg name="headless" value="$(arg headless)"/>
      <arg name="gui" value="$(arg gui)"/>
      <arg name="world_name" value="$(arg world)" />
      <arg name="debug" value="$(arg debug)" />
      <arg name="verbose" value="$(arg verbose)" />
      <arg name="paused" value="$(arg paused)" />
  </include>

  
<group ns="uav1">
    <arg name="name" default="iris_1"/>
    <arg name="ns" default="/iris_1/"/>
    <arg name="fcu_url" default="udp://:14540@localhost:14557"/>
    <arg name="vehicle" default="iris_1"/>
    <arg name="rcS" default="$(find px4)/posix-configs/SITL/init/$(arg est)/$(arg vehicle)"/>
    <arg name="sdf" default="$(find mavlink_sitl_gazebo)/models/$(arg vehicle)/$(arg vehicle).sdf"/>
    <arg name="gcs_url" value=""/>

    <arg name="x" default="0"/>
    <arg name="y" default="0"/>
    <arg name="z" default="0"/>
    <arg name="R" default="0"/>
    <arg name="P" default="0"/>
    <arg name="Y" default="0"/>

    <node name="$(anon vehicle_spawn)" output="screen" pkg="gazebo_ros" type="spawn_model"
        args="-sdf -file $(arg sdf) -model $(arg name) -x $(arg x) -y $(arg y) -z $(arg z) -R $(arg R) -P $(arg P) -Y $(arg Y)"/>

    <node name="sitl" pkg="px4" type="px4" output="screen"
            args="$(find px4) $(arg rcS)">
    </node>

    <include file="$(find px4)/launch/mavros.launch">
        <arg name="ns" value="$(arg ns)"/>
        <arg name="gcs_url" value=""/> 
        <arg name="fcu_url" value="$(arg fcu_url)"/>
        <arg name="pluginlists_yaml" value="$(arg pluginlists_yaml)" />
        <arg name="config_yaml" value="$(arg config_yaml)" />
    </include>
</group>

</launch>

'''

But I got a new error:
'''
[uav1/vehicle_spawn_xx_MS_7922_28787_8827596932026697940-4] process has died [pid 28851, exit code 1, cmd /opt/ros/kinetic/lib/gazebo_ros/spawn_model -sdf -file /home/xx/Firmware/Tools/sitl_gazebo/models/iris_1/iris_1.sdf -model iris_1 -x 0 -y 0 -z 0 -R 0 -P 0 -Y 0 __name:=vehicle_spawn_xx_MS_7922_28787_8827596932026697940 __log:=/home/xx/.ros/log/d44c0db6-67d9-11e7-9aba-4ccc6a27bd2a/uav1-vehicle_spawn_xx_MS_7922_28787_8827596932026697940-4.log].
log file: /home/xx/.ros/log/d44c0db6-67d9-11e7-9aba-4ccc6a27bd2a/uav1-vehicle_spawn_xx_MS_7922_28787_8827596932026697940-4*.log

'''

closed 2017-07-13 by gyang1011

ChenYunan, 2017-07-20:
@gyang1011 Hello, I have the same ERROR as yours. Did you solve this error? Why did you close this issue?

khris-kabbabe, 2017-08-03:
@gyang1011 Hi, I would like to know what you solution was. I followed the instructions but have not been able to get this working. Any help would be appreciated.

EdXian, 2017-08-12:
@khris-kabbabe  Hi  ,  i wonder know which line you removed in the file `single_vehcile_spawn.launch`
Dose the file  still work fine without "16 line" ?
Thanks!!

YCH188, 2018-02-04:
I found a solution,but haven't tried.Maybe it will help.
https://github.com/PX4/sitl_gazebo/pull/101

lamping7, 2018-02-13:
@YCH188 multi_uav_mavros_sitl.launch was fixed in https://github.com/PX4/Firmware/pull/8831 However, be aware that the dev guide needs updated

hamishwillee, 2018-02-13:
@lamping7 What changes need to be made in the devguide? Is there a tracking PR/issue?