PX4/PX4-Autopilot #7597: BUG: Code dependent on the compiler

ChaosSeeker, 2017-07-11:
Hey there.

I've been trying to run the SITL version on a rpi3 and when compiling with the arm-linux-gnueabihf-raspbian toolchain the simulation was not running correctly.

After a few days of hunting the bug down, I found that, when not defining a CAL_MAGX_ROT, it's value is left at -1, and this value is then cast to the Rotation enum, which starts at 0. This causes undefined and problematic behavior, especially with the aforementioned toolchain. It caused the rot matrix return to be different than expected, mag report values to drift, causing an increase in the innovation value of the ekf2 loop and signaling a fusion timeout, causing the estimated position to jump all over.

I understand that this behavior might not show up when the config file is properly done, or when working with a full fledged ground control software, or with the specific native toolchain, but in my bare boned setup where I'm developing an offboard controller and a ground control of my own, it makes a difference. For me I just set the defaults of all the CAL_MAGX_ROT to 0 on ~/src/Firmware/src/modules/sensors/sensor_params.c, but a more appropriate fix could be made for the official repo.

I've also been able to replicate the problem when compiling with gcc and running it on fc24.x86_64, mainly when printing several times in what should have been quick loops.

Don't know if anybody will see this, but from my perspective code should never be dependent on the used compiler.

PS: I'm a new developer and I'm sorry if I missed anything that would make this bug report useless. From my perspective, opening the issue seemed the right thing to do

dagar, 2017-07-11:
Those mag rotation parameters are intended for external modules. A value of -1 is used to indicate the mag is internal and not apply an additional rotation.

Can you describe the full situation? Maybe something else is going on here. Was this an internal or external mag? Which driver and where are you seeing the cast?

TSC21, 2017-08-20:
@Mtbarros we need more info on this so to help you out

Mtbarros, 2017-10-03:
Hey there  @TSC21 , sorry for being out of the picture for so long. I went back and checked and apparently this has been fixed already!

https://github.com/PX4/Firmware/commit/0cb3eb99dcfdb47ba722f0ea6798299456bd7aca#diff-a4442851925fd43c16a781c4b830eb87

As you can see, before that commit, the code would get mag_rot, and then immediately cast it to the Rotation enum. ((enum Rotation)mag_rot). As such, when a value of -1 was passed the behavior was undefined (I think). And when using a less mature compiler (arm-linux-gnueabihf-raspbian) it was doing a random rotation. (This was quite a headache to figure out, as my first symptom was simply flight instability while trying to run the simulation). Glad this has been fixed.

Mtbarros, 2017-10-03:
Also, my first post was after the fix, how ironic! I had a version prior and wasn't updating it. I will mark this as closed as it was already solved.

closed 2017-10-03 by Mtbarros