PX4/PX4-Autopilot #7599: `EKF GPS fusion timeout` errors with good GPS data available lead to big position estimate jumps

tstellanova, 2017-07-11:
While flying an octo with ublox M8T GPS we see rapid extreme position estimate jumps, as logged here:

http://logs.px4.io/plot_app?log=d6944989-20ef-4007-bf1e-ca0b3732347f

- Multiple warnings are logged similar to: `2:24:865 | WARNING | [lib__ecl] EKF GPS fusion timeout - reset to GPS`
- Have tried two different GPS units (ublox M8N, M8T) on several different airframes with the same results.
- Looking at the code it appears this warning shows up when ECL decides GPS data is now valid after having been invalid for some time. From what I can tell from the logs, a good GPS position was continuously available during operation. 
- This problem has not yet appeared using the UAVCAN Zubax GNSS2 (which does not use the ubx gps driver.)

Possibly related to:
- https://github.com/PX4/Firmware/issues/6443
- http://discuss.px4.io/t/position-estimate-problem-in-flight/3162/4

tstellanova, 2017-07-11:
-  Another airframe with similar results:
http://logs.px4.io/plot_app?log=f72f8bdb-372d-4c83-a4a8-b929406ec059

priseborough, 2017-07-12:
Here are some preliminary observations:

1) The measured magnetic field length is invariant through the orientation changes which makes a compass calibration an unlikely cause.
2) Vertical innovations are low
3) Z vibration is lower than X and Y.
3) GPS horizontal velocity innovations are large from the start of the log. This can be caused by aliasing of gyro data, clipping or aliasing of horizontal accelerometer or a magnetometer with an incorrect yaw orientation set.

Can you please:

1) With the vehicle outdoors verify the yaw orientation shown on QGC with the copter pointing towards a known landmark.
2) Ensure that the motor vibration frequencies do not cause the vibration isolation mount to resonate. The levels of high frequency vibration in the IMU data are higher than they should be.

dagar, 2017-07-12:
@tstellanova this is a bit of a long shot, but where is the external mag connected?

tstellanova, 2017-07-12:
- Will attempt Paul's recommendations in a day or two on at least two separate airframes. 
- @dagar I believe in all instances the external mag is connected via GPS2 on the Pixhawk2.1   .  Tested both Drotek M8T XL and Holybro M8N , which both include mag.

dagar, 2017-07-12:
Can you carefully check all axes of the mag? The pixhawk 2 used what the pixhawk 1 considered an internal i2c bus in the 2nd GPS port. It's supposed to be binary compatible with px4fmu-v2, but based on how these things are currently implemented it's really not. The common external gps + compass modules typically have the arrow aligned with -y so there's an assumed rotation if the module is external.

If this is the case the quick fix is to use GPS 1, and we'll discuss proper options with @davids5.

tstellanova, 2017-07-12:
We tested two airframes outdoors and found that the heading was exactly backwards, nose-tail (pointing the aircraft West yields a heading of East in QGC).   This does appear to be a bug with the mag calibration since we used two different GPS + mag modules. 

The cabling for GPS1 and GPS2 is slightly different so it'll take a little time to reconfigure for GPS1.

dagar, 2017-07-12:
You can hack it here to verify. https://github.com/PX4/Firmware/blob/master/src/drivers/hmc5883/hmc5883_i2c.cpp#L115

Then if you recalibrate the corresponding mag cal rotation parameter shouldn't be set to internal. You can also check with the listener command and look for the is_external field.

`listener sensor_mag 0 1`

mhkabir, 2017-07-12:
@tstellanova Are you on master?

tstellanova, 2017-07-12:
@mhkabir Essentially master, yes

- Tried recalibrating on GPS2 and confirmed that heading is exactly backwards after calibration
- Connected the same GPS+mag unit (Drotek M8T XL) to GPS1 and recalibrated.  The heading shown after calibration is correct.

mhkabir, 2017-07-12:
OK I see what's going on. In the first case, I2C2 is using `PX4_I2C_BUS_ONBOARD`, and therefore the mag driver thinks that it's an on-board mag, and applies the board rotation to it, instead of applying the external rotation.

dagar, 2017-07-12:
@mhkabir I've discussed with @davids5 and it seems like we probably need to stop treating the cube like it's just another px4fmu-v2.

I proposed using px4fmu-v3 to refer to the cube (both solo and 2.1) and for px4fmu-v2 have a default config limited to 1MB (original pixhawk), and a larger config with the additional modules (mRo pixhawk, pixhawk mini). The only problem with this is we can't tell the difference between a newer pixhawk and pixhawk cube at flashing. Ideas?

davids5, 2017-07-12:
Yes.

The problem is the px4fmu-v3_default is a px4fmu-v2_default build with only a linker map change to support the full flash because the assumption in that the CPU is >= rev 3 silicon and the bug is fixed

The bus internal/external changes have not been accounted for.

As @dagar says it is time to split them off. But we have the issue that the bootloader is reporting the same board id. So I think we have to do a run time board determination and find all the compile time dependancies and  and address them.

Siytek, 2017-08-27:
Not sure if its related but I just had a crash and after looking at the log I have the GPS fusion error. The GPS data seems good, the flight path looks correct but the estimation is way off. The plane flew great in stabilise, however once I got half a km out I went to test hold mode and about 90 degrees into the turn the plane went into panic land mode.

Post crash I rewired the electronics and checked the position in QGC. Everything seems correct, the plane points the correct way on the map and the compass heading seems correct.

I am running stable on PH1 hardware, any expert opinion would be greatly appreciated!

[image]

Log: https://logs.px4.io/plot_app?log=4215f0f4-5bc7-45cc-84d7-b7f7758f0091

Not the first time I had to re-glue the nose back on, a bit of packing tape and some glue, and she will be as good as new :) Only damaged equipment was smashed camera lens and broken VTX so could have been worse!

[image]

LorenzMeier, 2017-08-27:
Thanks for the report and sorry to hear! Here are a few first observations:

  * Your current measurement seems unusual: https://logs.px4.io/plot_app?log=4215f0f4-5bc7-45cc-84d7-b7f7758f0091#Nav-Power
  * Most likely a significant contributor: The acceleration here looks like it might be aliasing clipping some time into the flight: https://logs.px4.io/plot_app?log=4215f0f4-5bc7-45cc-84d7-b7f7758f0091#Nav-Raw-Acceleration

Overall the estimator reports a bunch of errors, so we definitely need to look at the setup you are flying and also check if it requires hardware changes, calibration fixes or if we can make the estimator more robust by adjusting the parameters for your plane.

Siytek, 2017-08-27:
Thanks for your comments Lorenz. Its no bother, you have to expect it from time to time! Its nearly end of season here anyway and I was planning on fibreglassing the nose over winter anyway. 

I just fitted a [Mauch current sensor](http://www.mauch-electronic.com), they are supplied with calibration values but this may need some tweaking. It was my first flight with this sensor so if the values are incorrect it wouldn't surprise me.

Any help would be much appreciated just let me know what you need to know regarding my setup. Ill probably rebuild it over the next couple of weeks and recalibrate everything. I have flown this airframe quite a few times on older revisions of the firmware without a problem, although I have made some changes to the internals as well as a jump in firmware revision since I last flew. The PH is now mounted on an acrylic plate that spans the battery bay, including 4 damper balls. This plate is probably more flexible than the previous fibreglass one so maybe there is some vibration issue?

priseborough, 2017-08-28:
The primary cause of GPS rejection was a 100 degree yaw error at takeoff. Look at the EKF vs GPS COG yaw:

[image]

EKF yaw innovation at launch was low so the EKF yaw was consistent with the magnetometer. The compass field strength is fairly constant throughout the flight which makes a bad calibration or electromagnetic interference less likely.

The most likely explanation is either the  compass being used is rotated by 90 degrees, a parameter has been set incorrectly or there is a FW bug in the rotation of compass data for that board or external compass module.

The vulnerability of fixed wing vehicles to misaligned compass units has been reduced in master, with the EKF now able to detect the misalignment at takeoff and reset the yaw to the correct value using the GPS and inertial data. The stable version you are using does not have this fix.

Siytek, 2017-08-28:
Wow outstanding evaluation, that's much appreciated! The PH is forward facing and unchanged from my setup over the last couple of years, however I moved the compass (GPS module) position prior to this flight and rewired it. As a result the directional arrow was pointing left by 90 degrees relative to the forward direction. The setting I used was 270 degree yaw rotation, is this correct? I did run recalibration prior to the flight as it was the first time I flew with the new physical mag position.

Turns out one evening with the glue gun and she's looking in much better shape now! I'll take it out for another flight soon, just need to replace the vtx which was due an upgrade anyway. I'll do a short manual flight first and check the log to confirm the estimate is good before testing any auto modes. Of course I think it would probably have been a good idea for me to do this on the last trip out seeing as I made so many major changes to the build. That said I do love FPV because of the learning curve/challenge, every crash is a blessing for the amount it teaches you!

priseborough, 2017-08-28:
Don't fly until you have resolved the yaw issue. The angle shown in QGC should match the real heading to within 15 deg.

Siytek, 2017-08-28:
Ok thanks for the advice, do really appreciate your feedback. Yes I'll ensure yaw is good, then I'll fly manual and check the log before trying any auto modes again.

LorenzMeier, 2017-12-31:
Closing as stale.

closed 2017-12-31 by LorenzMeier