PX4/PX4-Autopilot #7600: LPE: Can't get Local Position with PX4_Flow and Sonar

lvyanxuan, 2017-07-12:
I am using the new 1.6.4dev lpe firmware and want to achieve indoor navigation with PX4FLOW and Sonar. I am wondering which value the LPE_FUSHION should be set. If not fusing baro, the px4 seems does't know how to use the sonar data and can't get local position. btw, i can see the DISTANCE SENSOR(#132)and current distance seems work well.

TobiMiller, 2017-07-12:
so you have compiled the LPE firmware by yourself?

lvyanxuan, 2017-07-12:
@TobiMiller Yes, I compiled and uploaded it with "make px4fmu-v2_lpe upload".

jgoppert, 2017-07-16:
Did you try LPE_FAKE_ORIGIN=1.

TSC21, 2017-07-30:
@lvyanxuan updates?

lvyanxuan, 2017-07-31:
@TSC21 I have tried LPE_FAKE_ORIGIN=1. But it still can't get local position if not fusing Baro.

mindThomas, 2017-08-29:
I can confirm this with 1.6.5dev as well. No LPE estimate and therefor no access to ALTHOLD or POSCTL if Baro is not enabled.

TSC21, 2017-08-29:
Not considering this a fix, but could you guys enable baro and set a an high std dev for baro (`LPE_BAR_Z`)?

ecmnet, 2017-08-29:
That is exactly how I solved the issue for me.

TSC21, 2017-08-29:
> That is exactly how I solved the issue for me.

Well it is not ideal, but should work. I have to address this with @jgoppert so we don't have to have this kind of hacks.

closed 2017-08-29 by TSC21