PX4/PX4-Autopilot #7601: Strange behaviour in indoor offboard control with external positioning system

ccc159, 2017-07-12:
We are trying to fly our customized quadcopter in gps-denied indoor environment with mocap system.

At **first step**, we followed [gazebo SITL dev guide](https://dev.px4.io/en/simulation/gazebo.html) to simulate the setup.

In simulation we used **px4-1.6.3 Firmware**, **latest mavros**,  in ROS kinetic, Ubuntu 16.04.
The steps are:
1. ```make posix_sitl_lpe gazebo```
2. disable gps by ```gpssim stop``` to simulate gps denied environment.
3. start **mavros** and **Qgroundcontrol**
4. in **Qgroundcontrol** set ```LPE_FUSHION``` to 4.
5. run our own code to send mocap pose of drone from gezabo to ```mavros/mocap/pose```
6. in mavros we can see ```mavros/local_position/pose``` and ```mavros/mocap/pose``` are matching.
7. run our own code to send target point to ```mavros/setpoint_position_local```.
8. in **Qgroundcontrol** we can see mavlink message ```LOCAL_POSITION_NED ( #32 )``` and ```SET_POSITION_TARGET_LOCAL_NED ( #84 )```
9. in **Qgroundcontrol** we set flight mode to ```OFFBOARD`` and ```arm``` the drone.
10. the drone goes to destination in gazebo without problem.

Second step, we try to transfer this setup to real drone.
Hardware: 3dr pixhawk, RasPi 3, **No gps or flowsensor**.
Software: ```make px4fmu-v2_lpe upload``` Firmware, latest mavros, in ROS kinetic, Ubuntu Mate.
1. in **Qgroundcontrol** set ```LPE_FUSHION``` to 4.
2. run our own code to send mocap pose of drone from external positioning system to ```mavros/mocap/pose```
3. in mavros we can see ```mavros/local_position/pose``` and ```mavros/mocap/pose``` are matching.
4. run our own code to send target point to ```mavros/setpoint_position_local```.
5. in **Qgroundcontrol** we can see mavlink message ```LOCAL_POSITION_NED ( #32 )``` and ```SET_POSITION_TARGET_LOCAL_NED ( #84 )```
6.```arm``` the drone from flight controller in ```MANUAL``` flight mode, then switch to ```OFFBOARD```
7. drone starts to rotate, oscillate and somehow **didn't respond to our setpoint position at all**.

Here are two logs:

[first log we setpoint at ```[0, 0, 0.5]```](http://logs.px4.io/plot_app?log=dd51edcd-e268-4d8a-88de-d802aa40fafd)

[second log we setpoint at ```[0, 0, 0]```](http://logs.px4.io/plot_app?log=30ba35e2-d83a-4854-bc24-879ce16315ab)

but they both behave the same.

Since we are not able to read logs and get useful info. I would ask for help what problems are in our real drone setup.

THANKS!

ccc159, 2017-07-15:
Someone please react to this?

jgoppert, 2017-07-15:
Check that your mocap data makes sense and that the frame is correct. Mavros assumes you send it mocap data in enu frame typically and then rotates it to ned before sending it to lpe. Try doing a walk test with your mocap anf plotting local position to check that it is correct.

TSC21, 2017-08-07:
@ccc159 what's the status of this?

EdXian, 2017-08-11:
@ccc159 Do you make a progress in your problem?  Can you check the value of local_pose_z which should be consistent with your drone's mocap pose ? 
This is my problem : https://github.com/mavlink/mavros/issues/767

ccc159, 2017-08-11:
@TSC21 @EdXian hey guys sorry for my late apply. Since I'm just an architecture student without robotics background, after failing a lot of times trying offboard control following [dev guide](https://dev.px4.io/en/ros/external_position_estimation.html) without understanding what was happening frustrated me a lot, I decided to give up this method. 

My instinct told me the problem was the firmware, because around a year ago when I first tried pixhawk+px4fow in position control mode that was flying very well. **But with this new lpe firmware the drone couldn't even have a good position control flight**. Although everything is open source but to me it's like black box and I can't even figure out what happened.

I have only two months left to finish my thesis so I used a very stupid but however understandable way -- rcoverride.  I used my mocap data to control RC directly, mimicking a human controlling the controller. Although the flight now is quite unstable but at least it works and I'm happy with it...

Back to the issue, what I did the last time:

### Autopilot type and version ###

PX4 : px4fmu-v2_lpe
Firmware Version: 1.6.3

### px4 parameter setting ###

*LPE_FUSION : 12
*ATT_EXT_HDG_M : 2 (for mocap)
*Unchecked fuse barometer
*SYS_COMPANION : 921600

### Companion computer ###
raspberry pi 3
ROS: Kinetic
Ubuntu: 16.04
ttyACM0 : 921600

I could get mocap data consistent with localpose; I was sure the coordination systems are right according to the dev guide; I've done walk test and it was correct; I've made same setup in SITL and it was working.

But when I fly the real drone: desperation. It was unpredictable even under position control. As I said, I even tried without mocap but with flowsensor, it was still bad flight while flowsensor quality was high enough. 

I **guess** there's bug in new lpe firmware so I stopped testing anymore. Oh, probably my pixhawk is broken?

EdXian, 2017-08-11:
@ccc159 Thank you for your reply !! I suspect the problem is due to lpe algorithm in px4 firmware and your pixhawk is well. Why? Because I have tested atleast three pixhawk with `version:px4fmu-v2_lpe ` ,the problem still exist. I am a student with mechanical backgrond and I am newbie  in px4. Eventhough it is complicated and unfriendly to whom just get started in using px4 , I'm trying to learn how to use it.
If i  resolve the problem , i'll tell you . Hope that you can make a significant progress in your research .
Good Luck!!

closed 2018-02-21 by PX4BuildBot