PX4/PX4-Autopilot #7603: Battery Monitor not working on PixHawk Mini

gbrunso, 2017-07-12:
There seems to be a problem with the battery monitor on the pixhawk mini.  I calibrated the battery monitor, and set batt_low_thr to 20%, batt_crit_thr to 10%, and batt_energen_thr to 7% in QGC.  I managed to run a 4s battery down below 12V  without hearing any sort of alarm in flight.  I am flying indoors using offboard control and motion capture position information.
This happened using px4 1.6.3 v3_default firmware.

dagar, 2017-07-15:
Can you check the Power setup in QGroundControl? Make sure the the battery cell count is correct and check the calibration.

[image]

gbrunso, 2017-07-17:
Yes, all of my power setup and calibration is correct, except I did not change the amps per volt parameter because I do not have a good way to measure battery current.

dagar, 2017-07-17:
@davids5 did you try the power brick with your pixhawk mini?

gbrunso, 2017-07-17:
I am using the power distribution board that comes with the pixhawk mini.

[image]

davids5, 2017-07-17:
@dagar - tested on effectively master and the dual brick branch. 

Supply is set at 15.2V

master:
```
listener battery_status

TOPIC: battery_status instance 0 #1
timestamp: 35124820
voltage_v:  15.1946
voltage_filtered_v:  15.1821
current_a:   0.0000
current_filtered_a:   0.0000
discharged_mah:   0.0000
remaining:   1.0000
scale:   1.0000
cell_count: 3
connected: True
system_source: True
priority: 0
warning: 0

TOPIC: system_power instance 0 #1
timestamp: 97070274
voltage5V_v:   5.3641
voltage3V3_v:   0.0000
v3v3_valid: 0
usb_connected: 0
brick_valid: 1
usb_vaild: 0
servo_valid: 0
periph_5V_OC: 0
hipower_5V_OC: 0
```

Dual Brick:
```
TOPIC: system_power instance 0 #1
timestamp: 12620309
voltage5V_v:   5.3931
usb_connected: 0
brick_valid: 1
servo_valid: 0
periph_5V_OC: 0
hipower_5V_OC: 0
nsh> INFO  [lib__ecl] EKF aligned, (pressure height, IMU buf: 15, OBS buf: 14)
listener battery_status

TOPIC: battery_status instance 0 #1
timestamp: 25626882
voltage_v:  15.1372
voltage_filtered_v:  15.1809
current_a:   0.0000
current_filtered_a:   0.0000
discharged_mah:   0.0000
remaining:   1.0000
scale:   1.0000
cell_count: 3
connected: True
warning: 0
```

LorenzMeier, 2017-08-20:
Fine on master.

closed 2017-08-20 by LorenzMeier