PX4/PX4-Autopilot #7607: Pixracer external compass not detected

matt88, 2017-07-13:
Pixracer v1.0 on a new build using a ublox m8n gps compass combo. The gps works flawlessly?? The compass won't show up in calibration or anywhere in the parameters. The only two compass cal ids are known pixracer ids and unplugging the gps during calibration does nothing?  I have reflashed the fw. Using the latest stable 3.4.6 copter/hexa. I also had another m8n lying around. Tried it. No difference? The m8n I have said it was for 3.3 volts the pixracer has 5 on the gps port so I tried it on 3.3v. No difference. Have tried qgc,mission planner,and apm planner. No difference? Also tried a different memory card? I will try to get some logs to post if that would help? Tks in advance

LorenzMeier, 2017-07-13:
Please flash PX4 through QGroundControl if you want to use this software.

closed 2017-07-13 by LorenzMeier

matt88, 2017-07-14:
Tried that. Didn't make any difference. Still only showed internal compasses

shizzavip, 2018-02-19:
(matt88) did you ever get this worked out? I am having similar problem using x2.1.. Only internal compass is showing

matt88, 2018-02-20:
The seller ended up sending me a new board. It has worked great ever since. Only issue is no current sensing. I just ordered another pixracer from auav. Was hoping spending an extra 30 bucks would bypass any duds or flaws. Wrong! This one gets hot! And it keeps giving me bad ahrs warnings but it seems to work ok? Stick with pixhawk!

matt88, 2018-02-20:
Good luck finding help here btw. You read my first post and seen the case was immediately closed. I tried px4 fw. Even said I did initially but my case was closed at the first hint of a reason??

shizzavip, 2018-02-20:
I believe the really problem is the two groups went their own way, seems like a lot of bad blood between the two groups now. I have got into some pretty heated exchanges on the other side cause some folks on the ardupilot end go wide open when certain things are mentioned. Just this morning I was thinking what the hell im stuck in the middle of Betamax and vhs or blue ray and HD. And I was thinking maybe I should just say to hell with it all and just go to dji cause someone has got to lose and honestly I’m sick of feeling like it’s me every time I buy a dud ass pos device. You damn sure can’t get someone to give you a unbiased recommendation. You can google the shit out of something and still feel like you are just as lost as when you started. Personally I’m right back at square one on my flight controller search since my current x2.1 v2 needs to be crushed beneath Thor’s hammer. All the bs with pixhawk versions is just that bs. Normally a higher number means newer better. In this situation as far as I can tell it just the beginning and end of licensing rights. My understanding is your so quick to close the comment person is the proud pixhawk papa who has ended the license rights to others moving forward (just going off info read on forums please don’t kill me.. )anyway so now I’m stuck with do I do pixhawk 2.1 cube. Philip is supper smart and supper responsive to questions or maybe pixhawk pro 3 as this side is supper smart as well. All I want is something that works as stated and has good support.

matt88, 2018-02-20:
Before you give up check these guys out. This is where I bought my pixracer/food warmer however but they are already in the process of sending me a new one and seem to provide good support. They also have pixhawk and another fc that looks good. Auav.com

matt88, 2018-02-20:
https://store.mrobotics.io/category-s/107.htm

matt88, 2018-02-20:
I was not aware of the rivalry tho. Tks for that. That makes a lot of bs make sense now

shizzavip, 2018-02-20:
Auav has merged into mRo which was started by the half of 3DR (people behind the solo drone). I have bought almost all of my newer purchases from them and up till this last batch was happy then for oh say about 4 weeks no one would respond from support. Then yesterday one of the creators from auav Phillip himself jumped in to try and help out. I’m just kind of stuck. Currently I’m building a large hex but wanted to get very comfortable with my flight controller and other hardware so I purchased a decent size quad to work out any bugs. I would rather lose a shit frame with so so motors and stuff then have my larger hex with significant investment come crashing down.

shizzavip, 2018-02-20:
This is a little bit, but remember their is always 3 sides to the story so I’m not saying I believe this either. Lol I just want some shit that works. 🤬

https://discuss.ardupilot.org/t/ardupilot-and-dronecode/11295

dagar, 2018-02-20:
Hi guys, is there still an unresolved issue here? From the looks of it the issue was closed because it was an APM firmware issue, not PX4. If that's not the case feel free to reopen or create a new issue. Logs are always helpful and can be posted and shared via Flight Review (https://logs.px4.io/).

matt88, 2018-02-20:
My first pixracer sits on top of a tarot 680 pro frame with shortened cf booms and 920kv motors. I wanted a 550 sized hex with folding arms basically but if it helps u can fwd you the parameter file for it. I went with light weight components on everything btw. Very minimal plastic mostly carbon fiber. Auw is around 1700 kg if I remember correctly and that's with gimbal and GoPro. I think it measured out to be around 600mm instead of 550

matt88, 2018-02-20:
I get 10-12 minutes flight time on a 5500 4s lipo. I wanted 20min but I'll have to upgrade the motors for that and buy bigger batteries. I tried an 7000 mah bat but only netted around 13 min. For now it's just a toy for me to take pictures and video with. A very cool toy tho lol

shizzavip, 2018-02-20:
For me this is a issue... I’m using mRo x2.1 and I have followed the instructions for installing the correct bootloader as well. Under the compass section it shows only the internal no matter how many times I have installed the firmware through qgc. I’m trying to see if Phillip k. Can help out as he jumped in yesterday if not then I’m done on this one. The cheap pixhawk clone that I first bought didn’t give me this much trouble. No matter Ardu or px4. At this point I just trying to find enough info to help pull the trigger on the next purchase.

dagar, 2018-02-20:
@shizzavip can you create a new issue with a log attached? Alternatively you can ping me on PX4 slack (@dagar) http://slack.px4.io/ and we can go through some debugging.

Without arming you can get a log by flashing PX4 and setting SDLOG_MODE 2 to log from bootup to disarm.