PX4/PX4-Autopilot #7609: FW Land mode = crash

JonReacher, 2017-07-13:
I recently had a flight that put up a "[commander] failsafe: no global position warning", then switched to Land mode and proceeded to fly straight into the earth. My main concern is why the global position was lost in the first place (did the estimated position and gps position begin to differ too much? why?), but it would be nice to know that there is a reliable FW solution for this sort of condition to return to the ground hopefully in one piece.

Log here: http://logs.px4.io/plot_app?log=b7fcb8a9-a66f-4b49-94e3-f896d3a65cb7

JonReacher, 2017-07-13:
I know there are several ways that you could go about returning to the ground as safely as possible. In this particular case it seems that if the autopilot would have trusted the gps and disregarded position estimation everything would have been fine. If it truly is flying blind, or thinks it is, and no dead reckoning is available, would it be best to go into a descending loiter at a slow target speed, say FW_LND_AIRSPD_SC?

dagar, 2017-07-13:
There's an open loop loiter GPS failure mode, but I'd rather use the FW position controller to execute a heading hold landing while controlling airspeed. The main blocker to this at the moment is the FW position controller (TECS + L1) runs based on global position updates. I'm thinking of how to refactor it to continue running TECS.

Antiheavy, 2017-07-18:
A more typical loss of GPS failsafe in the world of fixed wing UAVs is to spiral down so as to hit the ground as close to the last reported GPS location as possible.  If you just heading hold you will likely get blown far downwind before hitting the ground - from 400 feet altitude a typical fixed wing might glide for over a mile downwind.

I'd vote for something like this:

1) Loss of GPS failsafe timeout ( 0.5 to 10 seconds depending on how long the given system can be expected to safely operate without GPS and/or without global position from the EKF).
2) Failsafe trigger
3) Set throttle = 0
4) Spiral down slowly holding airspeed: set to cruise speed or maybe minimum speed, hold a fixed bank angle
5) if GPS regained switch to Hold mode

dagar, 2017-07-18:
Thanks, I forgot about actually needing to find the plane after it safely lands.
If you were in a mission when you lost GPS and then it came back shouldn't the mission resume?

Antiheavy, 2017-07-18:
> If you were in a mission when you lost GPS and then it came back shouldn't the mission resume?

hmmm...  yeah, maybe you are right.   I'm thinking about if the vehicle descends down low and then decides to turn and start resuming the mission in some bad direction right into some trees or something, then again I don't know that Hold mode would be any better.   Less confusing to just resume whatever it was doing previously I suppose.

JonReacher, 2017-07-19:
I very much agree with spiraling down. Linear landing is scary for many reasons. A plane in a linear descent with a shallow glide angle can cover some serious ground before hitting who knows what.  At least with a descending loiter you have a reasonable expectation of the general "terrain" it will land in. The alternative may travel away from a safe flying area to the next town. Not joking, I fly on a ridge with a small city visible in the distance below a few miles away. One failsafe could set it on a course for real trouble.
  Is it too simple to just use the GPS failsafe navigation settings for the global position loss, except maybe with a scaled down throttle setting to ensure slow descent?  These should already be set for a holding loiter, at a reasonable radius for the specific aircraft.
  I'm no coder, but could you point to those parameters in case of a failsafe, then add a percent scale down, for example?

priseborough, 2017-07-31:
The current practice of fixed wing failsafe going into an open loop control mode is dangerous and often results in a high speed spiral dive, endangering people on the ground and providing little time for recovery. 

For fixed wing vehicles I would like to see addition of a CIRCLE_DESCENT mode that enables the plane to be put into a zero throttle, constant bank, constant pitch condition to provide a controlled slow speed spiral descent. If we assume that only the attitude solution is valid, then we cannot use TECS, so controlling to a specified pitch angle will be required. 

The only time the open loop mode should be used as the failsafe for fixed wing is when the attitude data in the control state message is invalid.

Antiheavy, 2017-07-31:
> If we assume that only the attitude solution is valid, then we cannot use TECS, so controlling to a specified pitch angle will be required.

2 cents: if valid airspeed (and baro altitude for that matter) are present they should be used instead of constant pitch.

> fixed wing failsafe going into an open loop control mode is dangerous and often results in a high speed spiral dive, endangering people on the ground

here is a recent fixed wing flight log where this exact spiral dive happened resulting in destruction of the vehicle:
http://review.px4.io/plot_app?log=1c1b7c69-d55c-4f63-a287-85d5be996453

priseborough, 2017-07-31:
The existing speed and height control loops (TECS) requires derivative states that are not available from using Baro or GPS directly, so until TECS algorithms are modified to generate this data internally, attitude control will have to do and will be a lot better than what we have currently.

priseborough, 2017-07-31:
I've been working on a progressive series of improvements to make the ekf faster to recover from a loss of navigation caused by bad magnetometer data. https://github.com/PX4/ecl/pull/294 a resulted in significant improvement in the ability to handle bad compass data and https://github.com/PX4/ecl/pull/313 will result in further improvements in that area when it has been tuned and tested.

The major remaining challenge is how to handle bad inertial data caused by excessive vibration levels. This can strike without warning in flight when RPM changes and structural resonances are excited and can make inertial navigation impossible.  What we need in that instance is a fallback mode that make minimal use of the accelerometer data. Until we disable the DLPF in the IMU and perform all digital processing in the driver, we will not be able to reliably detect clipping in the inertial sensors.

priseborough, 2017-07-31:
@Antiheavy Which version of the firmware were you using? I would like to know which version of the ecl library was being used.

priseborough, 2017-07-31:
I have replayed the 1c1b7c69 log file using both current  master and a version using  https://github.com/PX4/ecl/pull/313.

Velocity Innovations Before
[image]

Velocity Innovations After
[image]

The initial divergence which appears to be caused by a large magnetometer errors occurs in both instances, however the PR prevents the velocity states diverging again after the first GPs reset.

Antiheavy, 2017-07-31:
> @Antiheavy Which version of the firmware were you using?

i was a modified version of 1.6.3 or 1.6.4 - I don't think ECL was modified at all from those versions though.

dagar, 2017-07-31:
> Until we disable the DLPF in the IMU and perform all digital processing in the driver, we will not be able to reliably detect clipping in the inertial sensors.

If detected in flight is it recoverable?

priseborough, 2017-07-31:
If detected, we would configure the EKF to use an order of magnitude larger value for EKF2_ACC_NOISE and also larger values for all position and velocity innovation consistency check gates. It would significantly extend the amount of time and enable the estimator to survive either continuous light clipping or more intense shorter periods that can occur when resonances are excited by specific flight conditions and motor revs.

priseborough, 2017-08-22:
See https://github.com/PX4/Firmware/pull/7842

PR #7762 (2017-09-08): Navigator resurrect FW GPS failure navigation
GPS failure or loss of global position causes a transition to NAVIGATION_STATE_DESCEND, however this doesn't work for fixed wing as the position controller doesn't run without position updates.

 - fixes #7609
closed 2017-09-08 by dagar