PX4/PX4-Autopilot #7612: make: *** [px4fmu-v2_default] Error 2

bsowlin, 2017-07-13:
I tried to install px4 toolchains on Ubuntu 16.04 LTS. But when I used 'make px4fmu-v2_default' it shown like this
**[  0%] Built target git_nuttx
[  1%] Generating git hash header
[  1%] Built target git_genmsg
[  1%] Built target git_gencpp
[  1%] Built target git_matrix
Traceback (most recent call last):
  File "/home/bsowlin/src/Firmware/Tools/px_update_git_header.py", line 27, in <module>
    cwd='NuttX/nuttx', stderr=subprocess.STDOUT).decode('utf-8').strip().replace("nuttx-","v")
  File "/usr/lib/python2.7/subprocess.py", line 567, in check_output
    process = Popen(stdout=PIPE, *popenargs, **kwargs)
  File "/usr/lib/python2.7/subprocess.py", line 711, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1343, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory: 'NuttX/nuttx'
[  2%] Built target git_uavcan
CMakeFiles/ver_gen.dir/build.make:62: recipe for target 'build_git_version.h' failed
make[3]: *** [build_git_version.h] Error 1
CMakeFiles/Makefile2:1793: recipe for target 'CMakeFiles/ver_gen.dir/all' failed
make[2]: *** [CMakeFiles/ver_gen.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
[  2%] Built target git_mavlink
[  2%] Built target mixer_gen
[  2%] Built target nuttx_copy_px4io-v2
Makefile:160: recipe for target 'all' failed
make[1]: *** [all] Error 2
Makefile:164: recipe for target 'px4fmu-v2_default' failed
make: *** [px4fmu-v2_default] Error 2** 

How should I do?

dagar, 2017-07-13:
Can you try `make distclean` and then `make px4fmu-v2_default`?

bsowlin, 2017-07-13:
it's work. Thank you so much.
But can I ask for knowledge? what the purpose of make distclean?

dagar, 2017-07-13:
My guess is you didn't have the required submodules checked out. The build should have fetched and initialized them, so I'll have to take a closer look to see if something is broken.

`make distclean` is a nice way to completely clean the source tree, but also make sure all the submodules are in the correct state.

dagar, 2017-07-13:
I'll leave this open for now to see if I can recreate (and fix) and the failure.

LorenzMeier, 2017-08-20:
Stale.

closed 2017-08-20 by LorenzMeier