PX4/PX4-Autopilot #7614: Hi i use pixhawk with traxxas stampede(rover,rc car). How do i get car reverse??

Nickvot, 2017-07-14:
hi i use pixhawk with traxxas stampede(rover,rc car). 

but rover only forward

How do i get car reverse??

I use https://dev.px4.io/en/airframes_experimental/traxxas_stampede.html

dagar, 2017-07-15:
Rover support is quite new and started from the fixed wing controllers. We should look at adding support for this. Do you mainly want to use it in manual mode?

TSC21, 2017-07-30:
@marcozorzi maybe you want to advise here?

marcozorzi, 2017-07-31:
As far as I know the Traxxas esc has some sort of brake command that will be difficult to deal with in the px4 firmware. If you drive the rover with the original remote, accelerate and move the throttle lever all tje way back, the wheels will not spin backwards. They will act with a negative torque as a brake. Then, when you get to a full stop, move the throttle to zero and then back to negative throttle, it will go backwards. 
I did try to get it to go backwards (or even brake) but I failed to do that and moved on to something else.
My guess, if I were to try again, would be to try and edit the mixer and play around with the rc calibration so to get a +/- signal mapped to the output. (This would require, imho, to use a scope and double check what is coming out of the channel). 
Aftert this is achieved, then we could either keep the functionalities as the esc wants them or poke around with deeper level of the firmware to develop some sort of brake+reverse thing.

closed 2018-02-21 by PX4BuildBot