PX4/PX4-Autopilot #7615: Snapdragon flight spektrum not receiving

damul, 2017-07-14:
Hi, 

I cloned PX4/Firmware master branch.
I installed Hexagon SDK 3.0 and other tools with ATLFlight/cross-toolchain.

Building and uploading PX4 is successful with "make eagle_default" and "make eagle_default upload".
I can connect to PX4 with QGC, but radio is not responding. 

With deployed firmware from intrinsyc (3.1.3_PX4.zip from support.intrinsyc)
spektrum data was received.

What can I do for this problem?

r0gelion, 2017-07-14:
Hi @damul ,

When using eagle_default connect the receiver to J12 instead of J15.

damul, 2017-07-17:
Hi, @r0gelion.
Thank you for your answer. 
Is the pin map same as J15? (Is it Ok to just move connector to J12?)
Where can I find the information?

r0gelion, 2017-07-17:
Hi,

You can find the information [here](https://dev.px4.io/en/flight_controller/snapdragon_flight.html) at RC Wiring section

damul, 2017-07-18:
Thank you. I missed exact number.

closed 2017-07-18 by damul