PX4/PX4-Autopilot #7619: ROS launch scripts refer to overdefined executables

lachyd, 2017-07-15:
For example, in Firmware/launch/posix_sitl.launch we have:

<node name="sitl" pkg="px4" type="px4" output="screen"
        args="$(find px4) $(arg rcS)">
</node>

For me, under the Firmware directory there is a px4 executable python script (Firmware/Tools) plus another executable named px4 under the build target for a Raspberry Pi cross (Firmware/build_posix_rpi_cross). This is in addition to the px4 executable we want ROS to find (under Firmware/build_posix_sitl_default). 

If you run things in the right order when you set up your environment, ROS will know which executable you are talking about. If you get unlucky, ROS will prioritise one of the other executables. You can see the priority by typing 'rosrun px4 px4' and hitting enter. Unfortunately, roslaunch does not give the same prompt as rosrun. My temporary workaround is to remove execution permissions from the other two files.

The permanent fix is to put the px4 ros package under the build directory rather than the root Firmware directory. Some scripts will need to be modified for this to work.

anuppari, 2017-07-17:
This was noted in issue #7525 and has been fixed.

closed 2017-07-17 by TSC21

anuppari, 2017-07-17:
I should clarify. The overlapping names between the python script in Firmware/tools and the sitl in Firmware/build_posix_sitl_default was fixed by renaming the python script. If you're also cross-compiling on the same computer as running an sitl instance, then that's still an issue.

Temporary fix: Maybe just delete the RPi build after installing it on an RPi?

lachyd, 2017-07-17:
That's a possibility, but I prefer the permissions change mentioned above since it means that I don't have to rebuild the cross target every time. 

Long term I think the options are: 
- Copying the px4 sitl executable to px4_ros or some other name for sitl builds, so that ROS can find a unique executable; or
- Move the px4 ROS package to one level lower, into the build directory (preferred)