PX4/PX4-Autopilot #7623: Potential bug: Unreliable setting of home altitude leading to crash

jliphard, 2017-07-16:
Expected/desired behavior: vehicle boots and sets high quality home position (Lat, Lon, Alt), which is then used as reference point for mission, which then sets all altitudes relative to that home point. Correct setting of home altitude also important for RTL, since if the home altitude is wrong, RTL will try to land at that altitude. E.g. if the airfield elevation is 122m, and the home altitude is set to 0m, then the plane will fly into the ground when trying to RTL to 0m ASL. This is with QGroundControl 3.1.3, Pixracer, and PX4 VERSION: v1.6.5-116-g5618f34f3 -- CONFIG: nuttx-px4fmu-v4-default.

Context. My airfield is at 122m ASL. Sometimes, the home position altitude sets itself to 0m, and other times, it sets itself to 122m. Either number would make sense - the former is the relative altitude, and the latter is ASL. But why does it sometimes set 0m, and other times, 122m, and other times, the OSD shows --.--?

Potential Workaround. Let vehicle boot. Confirm location of home position. If alt = 0, or --.--, type in the real airfield altitude. I don't know if that is effective - probably not, since it's not clear to me that that number propagates into the Pixracer.

I'm enclosing screenshots of three different planned home positions at two locations. They should either be ~48m, or ~122m, but never 0 or --.--. As best as I can tell, this is not just a problem with the _planned_ home positions - this is reflective of the _actual_ home positions the plane is setting. I do not have a log for this flight, since I could not find the SD card in the debris, but I have one log from a few weeks back with a similar altitude setting problem (if that is useful) during switch to HOLD mode, where the plane dived upon STABILIZED->HOLD transition, but where i reacted quickly enough to switch back to STABILIZED and pull the plane up. 

[image]

dagar, 2017-07-16:
I think this is more of a planning issue with QGC, but we can verify exactly what's happening in detail.

In PX4 the HOME position is set initially when you first get a GPS fix and then updates anytime the vehicle goes from LANDED -> FLYING. This part we can verify in a PX4 log, or if you expand the mavlink inspector (under widgets) in QGC you'll see it under HOME_POSITION.

PX4 doesn't actually have a "Planned Home Position" and I think part of the confusion is the variability in the update of that box if you've never had a mission, opened a mission from file, or pulled the mission from the autopilot.

dagar, 2017-07-16:
Opened as a QGC issue - https://github.com/mavlink/qgroundcontrol/issues/5446

closed 2017-07-16 by dagar

jliphard, 2017-07-16:
I could not reproduce this problem. Subsequent [test flight](http://review.px4.io/plot_app?log=a598d080-779e-4b15-879b-82bcc6c56b23) with rebuilt airframe was error free.