PX4/PX4-Autopilot #7624: Flash PX4 onto Crazyflie 2 problem

ageofish, 2017-07-16:
I tried to flash PX4 onto my crazyflie 2, I tried master, stable and beta (via QGroundControl or manually), all of these finished flashing (erase, program, verify), but CF wasn’t able to connect with QGC. It seems the flie is still in boot loading mode (green M1 light blinks), QGC firmware tab detected the flie and suggest flashing:

Found device: PX4 FMU V2
connected to boot loader:
version: 5
Board ID: 12
Flash size: 1032192

I tried using the latest PX4 firmware, master, stable, and beta, none were able to connect after flashing.
On the other hand, I had a older piece of code “50ecee9d” Wed Jun 7 14:19:57 2017 that I can flash and connect with QGC, calibrate and so on.
Is this a known issue? Or did I miss something obvious?

[image]

[image]

YuhangWUST, 2017-07-19:
I have the same problem just like you. I'm Looking for the solution, and I haven't found it yet. Do you have any idea about this?

dagar, 2017-07-19:
@dennisss do you know anything about this?

YuhangWUST, 2017-07-19:
@ageofish I have already used the older piece code and succeeded flash and connect with QGC and so on, but cf2 cant connect with android QGC or official crazyflie phone app  and cant fly. Have you solved it?

ageofish, 2017-07-20:
Just saw the replies. Yes, I also found older commits on stable branch that works.
And I did a "binary search" on the commit history, it seems 1be089cf started to break the "stable" branch, maybe someone can take a look to see exactly why?

And what's interesting is that if I start with a working commit, and flash anything after 1be089cf, it still works until another certain commit. Then if I start with a non-working commit, flashing anything after 1be089cf does not work. Painful discovery, but hopefully someone with more knowledge will know exactly why given the hints.

dennisss, 2017-07-20:
I think the issue is right here: https://github.com/PX4/Firmware/commit/fc30f2906fd8b8dc5fe52c38a887e9e4e56b2e0a, some odd timing issue with the magnetometer resets

closed 2017-07-20 by davids5

YuhangWUST, 2017-07-20:
Thank for your help. It works. But there seems to be another problem, We use the latest firmware, can now connect to the computer side of the QGC ground station for debugging, but can not take off, and not connected to the mobile terminal QGC ground station, but the crazyflie official app can be connected, Between them seems to be able to communicate, because when I put the throttle down and the direction of the bar Right, M2, M3 bright blue light, and M1 extinguished, but it still can not take off. How can i solve this problem?

dennisss, 2017-07-20:
@YuhangWUST I see what you mean now. I've retested with a clean build and saw that one more little fix was needed. See https://github.com/PX4/Firmware/pull/7658

PR #7653 (2017-07-20): MPU9250 Reset reorganization (fixes I2C functionality)
During MPU9250 initialization, depending on passthrough modes, the main chip must be reset() before the magnetometer connection is initialized in order to setup the passthrough to the magnetometer.

Current sequence of events (during init):
---
...
- reset()
    - "MPU6050" reset
    - AK8963 reset <- fails because not yet connected
- Magnetometer init/connect

...

Fixed sequence:
---
...
- reset_mpu()
    - "MPU6050" reset
- magnetometer init
- ak8963_reset()
    - AK8963 reset

...

The functionality of `reset()` is preserved, but for t [... cut]