PX4/PX4-Autopilot #7625: Document SITL camera

LorenzMeier, 2017-07-16:
@dogmaphobic @DonLakeFlyer @dagar We have for several months now a pretty fancy SITL camera which is able to do these things:

  * Provide a real-time FPV video stream
  * Geo-tag images and store them on disk

It needs overall polishing and cleanup, but I think it is now at a point where we should invest into that + documentation. @hamishwillee @mrpollo I count on you for that.

The sketch below illustrates how this works and what layers are involved. I will send a set of PRs to docs and Firmware to make a couple of these things easier.

[image]

LorenzMeier, 2017-07-16:
Images:
[image]

Geotagging:

[image]

@Antiheavy this also works with the plane model and might give you ways to develop software and QA that will speed up your development exponentially.

LorenzMeier, 2017-07-16:
Couple more notes: Gstreamer for Mac OS:

  * https://gstreamer.freedesktop.org/data/pkg/osx/1.12.2/gstreamer-1.0-devel-1.12.2-x86_64.pkg
  * https://gstreamer.freedesktop.org/data/pkg/osx/1.12.2/gstreamer-1.0-1.12.2-x86_64.pkg

Flying a survey works:
[image]

Images get triggered and geotagged:
[image]

Last bit missing is some gimbal orientation debugging on this model, then its pretty much good to go.

LorenzMeier, 2017-07-16:
Ok, got this now under control properly:

[image]

[image]

[image]

LorenzMeier, 2017-07-16:
PR with the latest and greatest: https://github.com/PX4/Firmware/pull/7626

LorenzMeier, 2017-07-16:
Video streaming is working too, but QGC has an issue in recent versions displaying it: https://github.com/mavlink/qgroundcontrol/issues/5447

hamishwillee, 2017-07-17:
>  but I think it is now at a point where we should invest into that + documentation. @hamishwillee @mrpollo I count on you for that.

Sure. I'll look for your PRs to tell me you think it is time. Some information I will need/questions to answer.
1. Is the audience just people who want to use the feature, or people who may need to understand your code and develop this further (I assume the former)
2. Whether this works with HITL and if so how.
3. How you enable it in gazebo, SITL, builds
4. Where the images are stored.
5. Whether multiple cameras are supported
6. The diagram makes things pretty understandable, but I will probably need to clarify/confirm some of the interactions (for example, I don't know how GSC video works, so presume this is something I can just say "standard video connection" for).
7. What testing this now supports we couldn't before (e.g. surveys) - what else? Anything people might reasonably assume we can use it for that we can't?

closed 2018-02-21 by PX4BuildBot