PX4/PX4-Autopilot #7634: increase servo pwm 2150 us limit for vtol tilt rotor mechanisms

taileron, 2017-07-17:
I think it´s absolutely essential for full control and fun with the Convergence to have pwm signals that are only a small bit towards the original E-flite controllers values.
The original E-flite fc covers a (meassured) pwm range of 483 to 2560 us for the tilt rotor servos (ailerons 722 to 2322 us) thus I think a full safe (incl. symmetrical yaw) control of the Convergence with the PX4 default of 1000 to 2000 us is not really possible !! 
I know with pwm rates of 400Hz for the escs at 2350 us there are only 150us pause time left. Escs mostly don´t need more than 1950 us so it doesn´t care.
But the servo outputs work at 50 hz where a range of 500 .... 2500us would fit for most vtol purposes. The low limits can be successfully set with PX4 but currently there is a upper limit at 2150us which is still far away from real mechanical limits. If finally a limit is wanted it can be improved by pwm max.

dagar, 2017-12-06:
It's easy enough to bump the hard coded max, I'm just not sure if it could potentially damage someones setup?

taileron, 2017-12-06:
found it already in drv_pwm_output.h so you´re right it changes the whole behavior of mixers and their center positions and probably needs to tune up particular mixers but as long the default limit setting of 
 1000/2000 is active it should be harmless to increase the hardcoded absolute limit

dagar, 2017-12-06:
Are you able to make the change, compile, and test? If not, I could open a new branch.

taileron, 2017-12-06:
it works on a spare PixFalcon but now I have to change the mixers to fit again to the mechanics

taileron, 2018-01-09:
There is no problem if the hardcoded limit would be increased to a higher value (e.g. 2500) cause the default limits are used with default mixers. If in the sd´s extras file gives a higher limit an adjusted mixer (with geometric reduced values according the increased travel) is necessary anyway without needing to retune pid settings. Having an absolute upper limit at 2150 servo travel is mostly asymmetric and the center position can get out of alignment if the lower limit is changed below 850uS which is allowed.

taileron, 2018-07-04:
with all up to current firmwares 850 uS ..... 2150 uS can be set without changing the source by inserting an extras.txt into SD´s etc folder with the lines containing the desired limits (e.g. output 7,8):
pwm max -c 78 -p 2150
pwm min -c 78 -p 850
above 2150 uS needs to edit the source file "drv_pwm_output.h" PWM_HIGHEST_MAX from 2150 to 2500 which I think could be done into the released versions cause the default limits prevent all kind of malfunctions and if someone wants to increase pwms above 2150 that´s not blocked by the source.

DrDogMeat, 2018-07-06:
Wonderful thank you that worked a charm.

closed 2019-02-19 by stale[bot]