PX4/PX4-Autopilot #7635: The problem in pixhawk stay altitude.

lyhsteven, 2017-07-18:
Hey guys, I would like to let my pixhawk to stay altitude in Offboard Mode, so I connected a sonar MB1242 via IIC, and I upload the px4fmu-v2_lpe to hardware. But when I swich mode from manual to offboard, it seems not working. I wonder if I must switch mode from ALTITUDE to OFFBOARD, or I have to do something else.
Does anyone have the same issue that would help me,  I'll be appreciate to that.

lyhsteven, 2017-07-18:
Alao I want to do SLAM using vision, but the data of Z is not accurate, so I use a sonar to help stay altitude.

TobiMiller, 2017-07-21:
It is irrelevant from which flightmode you are coming. you need to publish a setpoint with const. z-pos. if it not hold its altitude your sonar is not working correctly. but a sonar isnt the most accurate sensor. a laser would do a quiet better job

lyhsteven, 2017-07-21:
@TobiMiller Thanks a lot. It worked! I published a target z-pos, and it would hold the altitude. It really helps a lot.

closed 2017-07-21 by lyhsteven

ZP-useful, 2017-11-28:
hello，i do the same thing with you, i also use mb1242 sonar to control the altitude, but when I witch mode to ALTITUDE ,the result is not good，the altitude sway up and down about 15cm, it not stay the goal altitude, Do you meet the problem before?Thank  you in advance!
this is my logs:https://logs.px4.io/plot_app?log=89c28b58-f256-4c96-a93f-2d4f053f96c9