PX4/PX4-Autopilot #7636: LPE errors and frequency issue Companion Link vs ESP8266 Mavros

crvogt, 2017-07-18:
Hi there, 
I'm trying to control a quad in a Vicon environment. I have an Odroid XU4 connected to a pixhawk via FTDI cable through telem2 and USB. 

I'm using firmware:  px4fmu-v2_lpe.px4

The issue is I'm struggling to get /mavros/local_position/pose at a reasonable rate to the Odroid.

The confusing part - 
I've chosen Companion Link (921600 baud, 8N1) from the dropdown menu in QGC. This leads to ~30Hz  data frequency from FCU, but /mavros/local_position/pose is not changing with the vicon data.
Instead, I'm getting a constant stream of _mocap timeout_ and _mocap position init_ with the correct vicon pose data. However, _/mavros/state_ is streaming the correct data.

When I choose ESP8266 (921600 baud, 8N1) (which I did out of desperation despite the previous setting being recommended), I get /mavros/local_position/pose CORRECT but at ~1Hz, and the output _lpe reinit P(0,0) negative_ and very few, but still some _mocap timeout_ warnings. /mavros/state output is correct here as well.

The USB to micro USB connection seems to work well, but I know that connection for data is discouraged.
In both cases Vicon data is being streamed at ~100Hz.

*sigh* I hope this was an adequate explanation. I'm at a big of a loss and thus struggling to describe what's happening. Any help is hugely appreciated :)

andrea-nisti, 2017-07-26:
How does the Odroid get vicon data?

blankered, 2017-08-01:
I think you need to publish to /mavros/mocap/pose. The topic /mavros/local_position/pose represents the published pose the UAV from the state estimators. If that is the case, let me further point out from my experience that the ESP8266 seems to struggle to work with streaming MOCAP data due to network latency issues. Suggestions have been made to reduce all other topic streams, modify streaming rates of data (requires 20hz I think), use regular telemetry or other wifi telemetry, or even play with UDP vs TCP (Multicasting or VRPN streaming). If you find a reasonable solution to use the ESP8266, please post what worked for you.

crvogt, 2017-08-01:
Hey, thanks for the input :) 
Unfortunately, no specific solution regarding the ESP8266
In my case turned out to be a timing issue more than anything, but I still don't understand the results. I ended up using the telem2 port and using vrpn_client_ros onboard the odroid. Prior to that, I had been using vicon_bridge on a laptop set as ROS_MASTER on the same network (ie Vicon) as the odroid, allowing them to share the data. The problem was that the timestamp for ROS on the laptop and for the odroid differed, whereas vrpn allows for ignoring the time on the vicon server.

closed 2017-08-01 by crvogt