PX4/PX4-Autopilot #7638: Change Altitude causes yaw setpoint change

jianyang1, 2017-07-18:
when loitering after reposition, changing the altitude will cause 180 deg change in yaw sp (facing the previous position)
 (FW v1.6.5 QG 3.2.0)

1. Take-off
2. re-position 100m north
3. when loitering after reaching the point- change altitude to 50m
4. bug- the drone changes the yaw setpoint (facing south)

http://review.px4.io/plot_app?log=a6cbe484-4ef9-46b4-91f2-f98e3708d407

TSC21, 2017-07-18:
Oh that's funny, as I'm getting the same behavior on SITL Gazebo. Maybe a different issue but totally worth to investigate. What's your vehicle setup?

BazookaJoe1900, 2017-08-11:
@LorenzMeier don't you think that this issue is priority?

VineeshKK, 2017-09-04:
Not able to replicate this issue. Tried to replicate it on Gazebo ver 7.8.1,  FW ver: 1.6.5 0 . QGC lateste ver.
This was the observation.
Initially Drone was directed towards East, then  it took off to 100m
After take-off , flew the drone towards 100m North.  (so the drone was directed towards North).
Loiter for 5 seconds once it reached 100m North.
Then reduced the altitude to 50m. 
I was still able to see the Drone directed towards North. which is expected
Did I miss any steps here?

jianyang1, 2017-09-04:
try start the maneuver in your takeoff altitude, then after flying north, climb to about 50 m.
i am still witness this behavior, in various maneuvers and altitudes, though it's not happening in **every** altitude change.
not sure whats trigger it but it is still happens pretty oftenly

BazookaJoe1900, 2018-02-17:
Any news about this issue?

dagar, 2018-02-17:
Please continue posting logs if you see this issue.

Stich123, 2018-02-22:
I have the same issue v1.7.3 QG 2.7.0
https://review.px4.io/plot_app?log=e1b133ba-a994-4aa6-868f-6893b505554f

RomanBapst, 2018-03-16:
I debugged this issue and found where the problem is and when the problem occurs.

The problem happens if you demand and altitude change if the vehicle has overshot the current position setpoint. As a result the navigator will calculate a new yaw setpoint which in that case is about 180 degrees off.
This is where the magic happens, as you can see there is a magic threshold of 1m.
https://github.com/PX4/Firmware/blob/master/src/modules/navigator/loiter.cpp#L150

@LorenzMeier once committed a fix to help with this https://github.com/PX4/Firmware/commit/e1e15add01bf9292bd636581037a8673b376af15

In the end in depends on vehicle tuning and the value of the magic threshold.

RomanBapst, 2018-03-20:
I'm closing this since this behavior is not really a bug but rather an effect of the position controller excessively overshooting the position setpoint (the threshold is set to 1 m).

closed 2018-03-20 by RomanBapst