PX4/PX4-Autopilot #7639: OFFBOARD mode horizontal velocity

lyhsteven, 2017-07-18:
I am now doing SLAM in offboard mode via vision, I wonder know if someone can help that which param is controlling the horizontal velocity in offboard mode. For my uav flys too fast to the target point.
I would so glad.

andrea-nisti, 2017-07-20:
Try this `MPC_XY_VEL_MA`

lyhsteven, 2017-07-20:
@andrea-nisti Thanks, I will test it.

lyhsteven, 2017-07-25:
Hey @andrea-nisti, I have tested the param `MPC_XY_VEL_MAX`, but It seems not work.

andrea-nisti, 2017-07-25:
Which value did you put and what velocity did the robot achieved?

lyhsteven, 2017-07-26:
@andrea-nisti I turned the velue `MPC_XY_VEL_MAX` from 8(default) to 1, and I use the UAV in offboard mode. When I gave a target point to the UAV, it reached the target very fast, I didn't know what happened.

andrea-nisti, 2017-07-26:
By very fast you mean more than 1 m/s ?

TSC21, 2017-08-13:
@lyhsteven is this solved to you?

lyhsteven, 2017-08-14:
@TSC21 I've tested it in my UAV, it worked in `MISSIOM MODE`, but still not in `OFFBOARD`. It may has some problems in my code, I will check it and then give an answer. So we can close the question first. 
Thx.

closed 2017-08-14 by lyhsteven