PX4/PX4-Autopilot #7647: Create virtual camera sensor for SITL testing

mcharleb, 2017-07-19:
The new camera capabilities need to be fully tested. The only practical way to do this would be with a virtual camera sensor that can exhaustively test the possible options. This also allows the options to be exercised on the QGC side as well.

mhkabir, 2017-07-19:
We already have support for a camera and gimbal in Gazebo SITL on the Typhoon H480 model.

mcharleb, 2017-07-19:
@mcabir Then we would need to extend it to support options being set from the new work Gus has been doing for the Camera API

LorenzMeier, 2017-07-19:
@mcharleb To do this we should create a camera base library. It should hook into here: https://github.com/PX4/sitl_gazebo/blob/master/src/gazebo_gst_camera_plugin.cpp

closed 2018-02-21 by PX4BuildBot