PX4/PX4-Autopilot #7648: Hardware and Sensors

mrpollo, 2017-07-19:
* ROS 1 serialization? CDR?
* What PX4 topics are bridged
* Would ROS support then depend on the ROS2 to ROS1 bridge?
* https://github.com/eProsima/Fast-RTPS
* https://github.com/eProsima/Fast-CDR
* https://github.com/eProsima/Firmware
* https://github.com/eProsima/Firmware/blob/micrortps/src/modules/micrortps_bridge/README.md
* Implement MAVROS over ROS 2.0
* Support conversion in a library that mavlink router and MAVROS can use (Marcus)

TSC21, 2017-07-19:
Hi guys. About this me @villi are going to sync to push a lib to MAVROS that will serve as a bridge between PX4 msgs coming on fast-RTPS and translating them into ROS. This was something that probably pass through you as usually I am not able to attend to the dev meetings at thaf time. Is there someone else I should sync with to get this done?
Also about MAVROS port to ROS 2.0 - that's also something on the maintainers TODO list.

TSC21, 2017-07-19:
Also to inform that Jaime Losa, eProsima CEO, is aware of this integration effort.

LorenzMeier, 2017-07-19:
I've mentioned this a couple times, just making sure it does not drop: I think we should put some effort behind making sure that mavlink-router (serving as bridge to FastRTPS) and mavros can share as much code as possible on the serialisation part. They are doing the same task for much of what they execute and they should base of the same base serialisation library.

TSC21, 2017-07-19:
I agree with you. But talking with the mavlink-router team the answer I got was short and something as close as "we are commited to mavlink-router at the moment". So unless we sync efforts to have a common library for fast-RTPS (that can support both MAVLink and CDR msg translation) to have a common base for both MAVROS and mavlik-router, we are goinf to end-up having different libraries doing the same. On a purpose-base perspective, it can also make sense as mavlink-router is a more general bridge than MAVROS, which focus on translation for the ROS framework, and based on that perspective @vilhjalmur89 agreed also that MAVROS could have a module implemented (as libmavconn) that could wrap CDR from and to fast-RTPS.

closed 2018-02-21 by PX4BuildBot