PX4/PX4-Autopilot #7651: Snapdragon PWM output performance

LorenzMeier, 2017-07-19:
We have reports of suboptimal performance with PWM outputs while the performance with the UART is good - can we audit the PWM output rate? It should be 400 Hz with 1-2 ms duty cycle (there is also OneShot, which we could enable later).

r0gelion, 2017-07-19:
I have 2 vehicles with Snapdragon.

1.- 200qx with Qualcomm ESCs (UART) - it is working, some issues maintaining altitude (slow response to RC).
2. QAV250 with SimonK ESCs (PWM) - The ESCs are not initializing. I loaded master and commit 5af62c8e, this last one used to work, not a good performance but the ESCs worked.

I checked the output with an oscilloscope and found the frequency is 400Hz but the positive with of the pulse is 4.23us and we need 1ms for the ESCs to initialize.
[image]

Using Flight BSP 3.1.3.1

LorenzMeier, 2017-07-20:
Thanks! @jywilson can you help to resolve this?

r0gelion, 2017-07-20:
I loaded eagle_legacy_default firmware to the board and confirmed the UART output is working, the problem is with PWM output (eagle_default).

r0gelion, 2017-07-20:
@LorenzMeier @jywilson 
My bad, I deleted the blsp.config file, once I uploaded it again it worked.

Thank you for your help!

LorenzMeier, 2017-07-21:
@r0gelion Can you re-test the flight performance with master? We improved a few things.

Avysuarez, 2017-07-24:
QAV 250 with snapdragon good flight, but has some issues to maintain altitude;
http://review.px4.io/plot_app?log=8c4ad17b-ef5f-4bfb-ad1a-b3e0ca0fa618

jywilson, 2017-08-22:
@r0gelion I am trying to catch up on this problem.  Are you still having issues with the pulse width?  The jitter that are seeing in the image of the scope above is expected since we are generating pulses in software with an unused timer.

r0gelion, 2017-08-24:
@jywilson That time it was my mistake, I just uploaded the blsp.config file and it worked, I uploaded master yesterday and the ESCs are not initializing, this time the blsp.config file is there. I'll try to find what is causing the issue this time.

r0gelion, 2017-08-26:
@jywilson we tried again with edea4a36 and it worked fine.
https://logs.px4.io/plot_app?log=48d1ac39-31a5-406e-a74f-fdc5a9bb7d71
https://logs.px4.io/plot_app?log=eb581178-28ee-4fe1-9ae9-7195d765b0b3
https://logs.px4.io/plot_app?log=348d5767-3c40-4e17-ae0d-d53a95d682b2

jywilson, 2017-08-28:
Okay, that's good news.  Thanks @r0gelion for the update.

jywilson, 2017-08-28:
@r0gelion I just checked the logs you reference above and they indicate that altitude does not change, even after moving the joystick to increase thrust.  Is this related to the PWM issues reported in this thread?

santiago3dr, 2017-08-29:
@jywilson this does not look to be directly linked tot he PWM issue as after a power cycle the copter fly just fine in the second and third logs reported; I am speculating as I could not scope the copter at the time of the full throttle not raising the copter in any mode, it lifted about 0.5 meters and stayed at that altitude until landing

ChristophTobler, 2018-01-10:
I guess this one can be closed. PWM ESC perform good on current master and can't be as good as the UART ones from Qualcomm...

closed 2018-01-10 by ChristophTobler