PX4/PX4-Autopilot #7655: PX4 build fails after including hc_sr04 driver in cmake

vmishra9, 2017-07-20:
PX4 build fails if drivers/hc_sr04 is included in the nuttx_px4fmu-v2_default.cmake

Following error is received during the build time:

error: 'GPIO_GPIO6_OUTPUT' was not declared in this scope
  {GPIO_GPIO6_OUTPUT,      GPIO_GPIO7_INPUT,       0},
   ^
compilation terminated due to -Wfatal-errors.
src/drivers/hc_sr04/CMakeFiles/drivers__hc_sr04.dir/build.make:62: recipe for target 'src/drivers/hc_sr04/CMakeFiles/drivers__hc_sr04.dir/hc_sr04.cpp.obj' failed
make[3]: *** [src/drivers/hc_sr04/CMakeFiles/drivers__hc_sr04.dir/hc_sr04.cpp.obj] Error 1
CMakeFiles/Makefile2:4615: recipe for target 'src/drivers/hc_sr04/CMakeFiles/drivers__hc_sr04.dir/all' failed
make[2]: *** [src/drivers/hc_sr04/CMakeFiles/drivers__hc_sr04.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....

yinjiajie, 2017-07-28:
@vmishra9 The hc_sr04 driver is not complete.

yinjiajie, 2017-07-28:
@vmishra9 You should use PWM to driver sonar.

closed 2018-01-17 by dagar