PX4/PX4-Autopilot #7656: ecl_ekf/process_logdata_ekf.py is broken

nicolaerosia, 2017-07-20:
Could someone please try the script and check if it is working? I get the following error:
```
./Tools/ecl_ekf/process_logdata_ekf.py /tmp/19_50_16.ulg 
found estimator_status data
found ekf2_innovation data
found sensor_preflight data
always on ground
always on ground
Traceback (most recent call last):
  File "./Tools/ecl_ekf/process_logdata_ekf.py", line 778, in <module>
    pp.savefig()
  File "/usr/lib64/python2.7/site-packages/matplotlib/backends/backend_pdf.py", line 2483, in savefig
    **kwargs)
  File "/usr/lib64/python2.7/site-packages/matplotlib/figure.py", line 1563, in savefig
    self.canvas.print_figure(*args, **kwargs)
  File "/usr/lib64/python2.7/site-packages/matplotlib/backend_bases.py", line 2232, in print_figure
    **kwargs)
  File "/usr/lib64/python2.7/site-packages/matplotlib/backends/backend_pdf.py", line 2536, in print_pdf
    self.figure.draw(renderer)
  File "/usr/lib64/python2.7/site-packages/matplotlib/artist.py", line 62, in draw_wrapper
    draw(artist, renderer, *args, **kwargs)
  File "/usr/lib64/python2.7/site-packages/matplotlib/figure.py", line 1159, in draw
    func(*args)
  File "/usr/lib64/python2.7/site-packages/matplotlib/artist.py", line 62, in draw_wrapper
    draw(artist, renderer, *args, **kwargs)
  File "/usr/lib64/python2.7/site-packages/matplotlib/axes/_base.py", line 2319, in draw
    a.draw(renderer)
  File "/usr/lib64/python2.7/site-packages/matplotlib/artist.py", line 62, in draw_wrapper
    draw(artist, renderer, *args, **kwargs)
  File "/usr/lib64/python2.7/site-packages/matplotlib/text.py", line 757, in draw
    raise ValueError("posx and posy should be finite values")
ValueError: posx and posy should be finite values
```

dagar, 2017-07-20:
The script is definitely still working with flight logs, but I've also seen this message in logs from ground tests more than a month ago. I didn't dig into the cause.

@priseborough FYI

nicolaerosia, 2017-07-20:
You mean error?

dagar, 2017-07-20:
Yes, this error and failure to process the log.

nkhoit, 2017-07-20:
This only happens if you process a log that doesn't involve taking off and landing. 
If you parse a log that involves actually flying around, it'll parse fine.

The problem is in_air_transition_time and on_ground_transition_time get set to NaN if the vehicle never takes off, but then it fails to print these plot texts: https://github.com/PX4/Firmware/blob/master/Tools/ecl_ekf/process_logdata_ekf.py#L758

closed 2018-02-21 by PX4BuildBot

kaklik, 2019-01-09:
Hi, the script behavior is still not fixed and fails on processing ground-only log files. 

```
  kaklik@k:~/repos/TF/PX4Firmware/Tools/ecl_ekf$ python process_logdata_ekf.py /home/kaklik/repos  /TF/PX4Firmware/build/px4_sitl_default/tmp/rootfs/log/2019-01-09/14_51_35.ulg
  Using test criteria loaded from /home/kaklik/repos/TF/PX4Firmware/Tools/ecl_ekf/check_level_dict.csv
  always on ground
  always on ground
/home/kaklik/repos/TF/PX4Firmware/Tools/ecl_ekf/analyse_logdata_ekf.py:1019: RuntimeWarning: invalid value encountered in greater
  start_index = np.amin(np.where(status_time > in_air_transition_time))
Traceback (most recent call last):
  File "process_logdata_ekf.py", line 83, in <module>
    late_start_early_ending=not args.no_sensor_safety_margin)
  File "/home/kaklik/repos/TF/PX4Firmware/Tools/ecl_ekf/analyse_logdata_ekf.py", line 1019, in analyse_ekf
    start_index = np.amin(np.where(status_time > in_air_transition_time))
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 2372, in amin
    out=out, **kwargs)
  File "/usr/lib/python2.7/dist-packages/numpy/core/_methods.py", line 29, in _amin
    return umr_minimum(a, axis, None, out, keepdims)
ValueError: zero-size array to reduction operation minimum which has no identity
kaklik@k:~/repos/TF/PX4Firmware/Tools/ecl_ekf$ 

```

kaklik, 2019-01-10:
The issue has been fixed in this commit: https://github.com/ThunderFly-aerospace/PX4Firmware/commit/8d9d1b9cc2a351d50905c0615dd4695bbe2599a3