PX4/PX4-Autopilot #7661: Compilation Error

ankushdesai, 2017-07-20:
I am getting the following error when trying to compile the Firmware:
[  0%] Built target prebuild_targets
[  0%] Generating ../../../logs
[  1%] Built target git_uavcan
[  1%] Built target git_matrix
[  1%] Built target logs_symlink
[  1%] Built target git_gencpp
[  1%] Built target git_genmsg
[  1%] Built target git_mavlink
[  1%] Built target git_jmavsim
[  1%] Built target git_driverframework
[  1%] Built target git_gazebo
[  1%] Building CXX object src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/DriverFramework.cpp.o
c++: error: unrecognized command line option ‘-fcolor-diagnostics’
src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/build.make:62: recipe for target 'src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/DriverFramework.cpp.o' failed

I have used PX4 Firmware extensively until few months back and this seems to be a recent change.

Any help would be appreciated.

bkueng, 2017-07-21:
Which compiler & version do you use?
The option is set here: https://github.com/PX4/Firmware/blob/master/cmake/common/px4_base.cmake#L409

ankushdesai, 2017-07-21:
Thanks for the response, I set the default compiler to gcc, g++ and it worked.

closed 2017-07-21 by ankushdesai