PX4/PX4-Autopilot #7664: Maxbotic sonar sensor not working

L4ncelot, 2017-07-21:
Hi,

I would like to use maxbotic MB1240 sonar sensor with px4, but unfortunately it's not working.

I've connected the sonar according to [this](http://ardupilot.org/copter/docs/common-rangefinder-maxbotix-analog.html) guide. I hear the sonar, so it should work. Then I uncommented line **#drivers/mb12xx** in **nuttx_px4fmu-v2_default.cmake** file. After that I just compiled the whole thing with `make px4fmu-v2_default` command and uploaded it with `make px4fmu-v2_default upload`. Finally I enabled the **SENS_EN_MB12XX** parameter.

But when I try to display the **distance_sensor** message in MAVLink inspector, I can't see it.

I'm using px4 firmware v1.6.5 with QGroundControl v3.2.1.

Does anybody know what's going on? Did I something wrong or am I missing something?

Thank you for answer.

andresR8, 2017-07-21:
@L4ncelot  Try with the QGroundControl daily builds.  I've done the same procedure as you and It's working with QGC 2017-07-20.

L4ncelot, 2017-07-24:
@andresR8, so I tried the newest daily build (2017-07-22) and nothing really changed for me. I did the same procedure and still can't see any distance messages. I have really no idea what's going on... the driver is there, the connection is correct and it's enabled in QGroundControl... I can try another release version. But other than that I don't really know what else to try.

L4ncelot, 2017-07-24:
Another weird thing is that according to [this](https://dev.px4.io/en/tutorials/tutorial_hello_sky.html) tutorial I should be able to start **mb12xx** driver from mavlink console by typing `mb12xx`. But this returns `ERROR [mb12xx] unrecognized command`. Although when typing `help` I can see **mb12xx** in builtin apps.

L4ncelot, 2017-07-24:
Ok so I've finally figured out where the driver fails. It fails right here during initialization of I2C:
```
if (I2C::init() != OK) {
		return ret;
}
```

My sonar is analog, therefore it's connected to ADC, not I2C.

mhkabir, 2017-07-25:
You will need to write your own ADC backend for the current driver. Please do send in a PR if you get around to it. 
I'm closing this for now since it isn't specifically a flightstack bug.

L4ncelot, 2017-07-25:
@mhkabir can you please point to some another analog driver which is already implemented to give me some direction how to approach the problem? I've never written hardware driver, so basically I'm not sure where to begin.

TSC21, 2017-07-25:
@L4ncelot you should check how other sensors that use serial interface are implemented. Check for example the [Aeroantenna μLanding driver](https://github.com/PX4/Firmware/blob/master/src/drivers/ulanding/ulanding.cpp).

closed 2017-07-25 by mhkabir

sivanagasaikrishna, 2018-11-16:
sir please help me where we need to check the sensor data in q gruound controller. is it by mavlink console .if it mavlink console whats that command ...please help me i am in learning stage.

hamishwillee, 2018-11-18:
@sivanagasaikrishna Please raise this question in the [support channels](https://docs.px4.io/en/#support).

IF there is some issue that you think might be related to your problem you can link to it from the support request at the above links. Please never just add a question to the end of a closed issue.