PX4/PX4-Autopilot #7672: Syslink can't handle >30 byte Mavlink messages?

ageofish, 2017-07-24:
New to Crazyflie, I have a 2.0 file and I've been trying to get Mavlink messages over Crazyflie PA for a few days.
Here's what I found via trial and error:
1. messages <= 28 bytes are transmitted flawlessly (e.g. SYSTEM_TIME, SERVO_OUTPUT_RAW, ATTITUDE)
2. messages >=31 bytes are received as random garbage (e.g. HIGHRES_IMU, ALTITUDE:
sysid:1 compid:1 time_usec:72506465346777910 altitude_monotonic:-1.045921E38 altitude_amsl:2.3803417E-38 altitude_local:-7.7662487E37 altitude_relative:2.3804148E-38 altitude_terrain:2.3693253E-38 bottom_clearance:1.2066013E-23')

Sending end is PX4 latest master running on Crazyflie2, receiving end is a Java program listening on port 14550, using DatagramSocket receiving DatagramPacket 1024 bytes at a time, and parsed with Mavlink generated java library, using PX4's v2.0/common.xml, protocol version 1.0.

I am guessing it's a syslink issue because when I run PX4 on simulator, all messages are received correctly, and simulator it does not go through Syslink.

I've been banging my head for a few days, any help would be highly appreciated!

mhkabir, 2017-07-25:
@dennisss, it would be good if you could have a look here.

dennisss, 2017-07-25:
@ageofish , the Crazyradio does bottleneck it at a well known 31 bytes per packet in hardware, although the driver in PX4 on the transmitting side should spread large mavlink messages across multiple network packets.

It should be noted that all mavlink messages are send with 8 bytes of header including a CRC at the end, so the ATTITUDE message is actually 36 bytes transmitted, so must be split into two packets.

Considering that both mavlink (per message) and the NRF radio (per packet) have error detection checks, it would lead me to believe that the messages were transmitted as garbage. Can you please confirm that you are getting good data from the Crazyflie via USB in QGC?

As a general rule, I usually keep the number of transmitted messages as low a rate as possible as the radio is very likely to drop messages once the buffers get full as it can not handle the datarate of the average PX4 transceiver.

ageofish, 2017-07-25:
How do I confirm "getting good data from the Crazyflie via USB in QGC?"
When CF is connected to QGC via radio, I didn't see any error messages, but nor was I able to see the actual messages -- I don't know where to look for them, and the MAVLink log file was empty (ulg files are created but no data was written).

dennisss, 2017-07-26:
QGC has a tool under 'Widgets -> Mavlink Inspector' which will show a rolling list of the messages received. Please compare the corresponding messages when using the radio. Do note that the USB interface is configured to send many more messages than the default configuration for the radio so only some of them will be comparable. You can also use this with QGC connected via the radio to verify that the issue is not with the Java parser

ageofish, 2017-07-27:
OK, I tested on USB all the messages are received as expected. As for radio, I don't have the hardware with me these two days, will report back.

ageofish, 2017-07-29:
I confirmed on QGC over syslink, long messages are lost.
What I see in QGC over radio:
HEARTBEAT, SYSTEM_TIME, ATTITUDE, SERVO_OUTPUT_RAW, VRF_HUD, EXTENDED_SYS_STATE

Compare to QGC over USB:
HEARTBEAT, SYS_STATUS, ATTITUDE, ATTITUDE_QUATERNION, LOCAL_POSITION_NED, HIGHRES_IMU and so on...

It seems it's a bug.

dennisss, 2017-08-04:
@ageofish sorry for the late response. So to clarify on the issue, are you receiving corrupted messages or are you not receiving all the ones you expect to be received. The radio interface is configured in OSD mode and only these (https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_main.cpp#L2089) messages are supposed to be send over the radio. Have you modified it or changed the mode to send a different set of messages?

ageofish, 2017-08-25:
Sorry for the late reply, I am receiving corrupted messages (I was able to print out the msgid and size, but the content was rubbish).
I modified the code to send HIGHRES_IMU in the OSD mode.
I am able to receive message less than 30 bytes, but rubbish if message is larger than 30.
Hope this clarifies!

closed 2018-02-21 by PX4BuildBot