PX4/PX4-Autopilot #7677: Main LED stays dark for VTOL configs after boot

AndreasAntener, 2017-07-24:
@bkueng I traced this back to https://github.com/PX4/Firmware/pull/6657 but I didn't yet investigate what the actual cause might be.

If the vtol attitude controller is sopped (`vtol_att_control stop`) via the LED starts to breath again.

bkueng, 2017-07-25:
I can repro. It's an excessive `status_changed` update in commander: https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L2135, causing LED resets at ~80Hz.
Can I leave that to you, I'm sure you know better which cases really require an updated status?

sanderux, 2017-07-25:
this? https://github.com/PX4/Firmware/blob/master/src/modules/vtol_att_control/vtol_att_control_main.cpp#L682

closed 2018-02-21 by PX4BuildBot